slipx06/sunsynk-power-flow-card #7: Charge / Discharge "dot" direction and Battery Runtime

Zodiac-69, 2023-04-29:
Hi slipx06

I have been using your Sunsync card for a while and it is GR8.
On the old card, i had to change the direction of the "dot" to show the correct direction of flow when my battery charge / discharge.
On this card, am not having the success. How do i "Swap" my setting to show the correct direction and the battery runtime?
I also noted on the non-essential load that there is a dot flowing from the load to the grid in the wrong direction.
Please point me to where i can correct this.
See attached image

https://user-images.githubusercontent.com/2076401/235299512-b8c5fd13-3ec8-4277-b4c4-48edc491908f.mp4

slipx06, 2023-04-29:
I've just pushed a fix for the power flow for the non-essential dot. How is your battery sensor configured. By default negative battery power is power flowing into the battery and visa versa

Zodiac-69, 2023-04-29:
I have it the other way around it seems.
If you look at the shot clip, battery is on float (13W) and i show positive.

slipx06, 2023-04-29:
The battery power displayed on the card will always be positive as we multiply any negative values by -1 (See code below). What is the raw sensor value shown in Home Assistant?

`<text id="battery_out_190" x="9%" y="92%" class="st2 st4 st8">${stateObj13.state < '0' ? stateObj13.state *-1 : stateObj13.state} W</text>`

The direction of flow as indicated by the animation is determined by positive or negative battery power.  Assuming you have one sensor that reports power leaving as a negative number and power flowing into the battery as a positive number.

Zodiac-69, 2023-04-29:
As it is now with the battery on float, HA is showing +16

slipx06, 2023-04-29:
Then its correct. Current is leaving the battery and the dot should move from the battery to the inverter?

Zodiac-69, 2023-04-29:
I don't have load shedding and the dot is moving from the battery to the inverter, HA show +16W.
See attached video above.
To me the dot should flow from the inverter to the battery, +16W - Float charging battery (power into battery).
When i have load shedding, my value then change to -xxxW - Battery discharging(power out of battery)

In Node-RED, this is my "decoding" of reg. 190 - From "Bloubul7" flow on powerforum.co.za

`
var a
a = (msg.payload.data / 1)
if (a > 32767) {
  msg.payload = (0 - a + 65535) / 1;
} else {
    msg.payload = (0 - a) / 1;
}
if (msg.topic === "battery_output_power") {
   return [ msg, null ];
} else {
   return [ null, msg ];
}
return msg;
`

slipx06, 2023-04-29:
The function looks correct but your values are being inverted somewhere. Are you sure you are not applying any other logic or there is something else in the flow that inverts this. Maybe attach a debug node.

If you have load shedding then the HA sensor should show a positive value i.e. +830W. This is what the card is expecting when drawing from the battery.

With my grid disconnected my battery power is positive.

[image]

[image]

You can edit the code for your unique case or fix the sensor. Change `< '0'` to `>'0'` and `>='0'` to `<='0' . You will need to find and replace this code for all 3 card styles

`<path id="bat-line" d="M 155 285.06 L 105.5 285.09 Q 95.5 285.1 95.5 290.92 L 95.5 296.75" fill="none" stroke="#f3b1c9" stroke-width="2" stroke-miterlimit="10" stroke-dasharray="2 2" pointer-events="stroke"/>
          <circle id="power-dot-charge" cx="0" cy="0" r="3" fill="${parseInt(stateObj13.state) < '0' ? 'transparent' : '#f3b1c9'}">
            <animateMotion dur="6s" repeatCount="indefinite" keyPoints="1;0" keyTimes="0;1" calcMode="linear">
              <mpath xlink:href="#bat-line"/>
            </animateMotion>
          </circle>
          <circle id="power-dot-discharge" cx="0" cy="0" r="3" fill="${parseInt(stateObj13.state) >= '0' ? 'transparent' : '#f3b1c9'}">
            <animateMotion dur="6s" repeatCount="indefinite" keyPoints="0;1" keyTimes="0;1" calcMode="linear">
              <mpath xlink:href="#bat-line"/>
            </animateMotion>
          </circle>`

Zodiac-69, 2023-05-01:
Hi slipx06
This is how i read the value from mqtt in my config file:
- name: "Battery Output Power"
   unique_id: "battery_out_190"
   state_topic: "SunSynk/battery_output_power"

Both mqtt and HA show positive value when battery is charging and a - value during loadshedding.
I don't have any other calculation / manipulation of the value once it is in HA.
All other Sunsynk values are read the same way from mqtt.

I have no idea why only the battery output power is "inverted"
[image]

closed 2023-05-05 by Zodiac-69