slipx06/sunsynk-power-flow-card #10: Resize canvas easily?

jacauc, 2023-05-02:
Hi @slipx06 

Do you know of a simple way to resize the canvas for the "lite" view with no solar installed so that is doesn't show blank whitespace at the top of the card?
Not a huge deal, just noticed it and thought it would be nice if the card was sized according to the contents.

[image]

slipx06, 2023-05-02:
I don't think it can be done easily as I think the position of the SVG elements are relative to the canvas and they will all move out of alignment. I will take a look. Another option is to fill that space with some text i.e. card title or other information?

slipx06, 2023-05-02:
I dont want to clutter the card but we can also extract the system timer info and display it 

[image]

slipx06, 2023-05-02:
I can centre the svg elements 

[image]

slipx06, 2023-05-02:
I managed to figure it out. I'll push to master 

[image]

jacauc, 2023-05-03:
Stepped out for a couple of days. This is perfect thank you!

closed 2023-05-03 by jacauc