slipx06/sunsynk-power-flow-card #17: 16kW Sunsynk has 3 x MPPT's and how can we add in Micro Inverter

starrateit, 2023-05-08:
Hi,

Any chance we could get an updated card which includes 3 x MPPT's?
Then can we use the Aux for power coming in from a Micro Inverter?

Thanks
Spencer

slipx06, 2023-05-08:
Hi. This will require some work to move svg elements around to make space. If there's enough demand for this it can be done.

starrateit, 2023-05-09:
Thank you, the same issue will exist with the larger 3 phase units they have 4 MPPT’s. So ideally design to encompass all versions, the inclusion of all 4 will be best and use case will increase!

Thanks,
Spencer

Use Mimecast to eliminate the risk of downtime and data loss during the migration to Exchange 2010 - for free through the <a href="http://www.mimecast.com/How-we-help/exchange-migration/#migration-assist">Migration Assist special</a>.

slipx06, 2023-05-09:
I have an idea on how we can add up to 4. :)

slipx06, 2023-05-09:
Can you reattach the image

starrateit, 2023-05-09:
[image]

starrateit, 2023-05-09:
Sorry there we go

slipx06, 2023-05-09:
Ok so that is 307 Watt input on the aux port. I'll check the logic. Dot should be moving from right to left

starrateit, 2023-05-09:
[image]

So the total for example should be this. if you want to do the maths and check everything

starrateit, 2023-05-09:
[image]

Here are the directions... 
I think Aux should be the other way based on the way the PV DC Solar is flowing

starrateit, 2023-05-09:
[image]

This gives you all the PV info including the Aux in the Entity tile on the bottom right

slipx06, 2023-05-09:
The thing is you can use the aux port feed power to another load or receive as you have i.e generator or micro-invetrer.  I need to create a config option to cater for this. In the mean time  you can create a template sensor to assign a negative value to input power to correct the flow. 

`sensor:
  - platform: template
    sensors:
      aux_inverted:
	friendly_name: "Aux Input Power Inverted"
        value_template: "{{ -1 * states('sensor.ss_aux_power') | float }}"
        unit_of_measurement: 'W' `

Thanks for the valuable feedback

starrateit, 2023-05-09:
Ok cool yeah that was what I was thinking, will do. I can also try and work around the MPPT’s for now by grouping. Not ideal as I can’t then see a problematic MPPT but will use it is a temp fix which should work.
Thanks for the help and your cool card. 😊

slipx06, 2023-05-09:
Something like this 

[image]

starrateit, 2023-05-10:
😀 yes that looks great 👍🏻👍🏻

[image]<https://user-images.githubusercontent.com/7227275/237191161-c93443d1-4aed-4ab8-91ea-d0b47ec01725.png>

slipx06, 2023-05-10:
I've just pushed a new release. v1.3.0 with this. There is also the option to invert AUX

starrateit, 2023-05-10:
Excellent I will give it a try!

starrateit, 2023-05-10:
[image]

Looking good, thanks for adding the option to remove the Non-essentials! 
Just a small bug seems to have come in where the flow direction has stopped working on the Grid and on the Aux. Or do I need to reload?

slipx06, 2023-05-10:
I'll fix that. I know what the issue is

slipx06, 2023-05-10:
Just pushed and update. Should be fixed in v1.3.2. Note that the AUX value of the card will not show a negative number. The flow animation is used to indicate input or output. You just have to make sure that your sensor records a negative number for input and positive for output

starrateit, 2023-05-10:
Cool thanks.

starrateit, 2023-05-10:
Ok will do a template for it and let you know all is sorted.

closed 2023-05-17 by slipx06