slipx06/sunsynk-power-flow-card #19: Essential / non essential issue

pawellygan, 2023-05-10:
I think I have found one more problem with essential and non essential  measurments.

Essential is going to UPS which is connected to LOAD in the inverter.

My aux is deactivated as I don't need aux. I use UPS (in case of power cut essential load will stay powered from battery)

It how it looks from an integration point of view:
[image]

now let's take your calculation for Essential load:

"essential = inverter_out_175 - (aux_power_166 - inverter_load_grid_169 )"

X = 500 - ( 0 - 500)
X = 500 + 500
X=1000

That is wrong.

[image]

My Essential load now is 0W in reality

slipx06, 2023-05-10:
You need to check which modbus rgisters Solar Assistant is using. Have a look at the inverter screen below.  The card is based on this logic. The only change I made was update 167 with 169 = 167+168.  inverter_out_175 is not the same as inverter_load_grid_169. They are different measurement points. 

[image]
[image]

slipx06, 2023-05-10:
Yes but remember that register 175 can be both positive or negative i.e. when charging batteries

[image]

pawellygan, 2023-05-10:
I'm still trying to understand that. I wrote email to SA with a question what is GRID LD

pawellygan, 2023-05-10:
Ok it looks like SA is doing this inside - it has a sensor NON-ESSENTIAL-LOAD

[image]

pawellygan, 2023-05-10:
that means when I use SA in between SunSynk and HA I can't use your card as

essential:
[image]

and non-essential:

[image]

Are exposed straight from SA as a separate entities.
The only option is somehow to skip your attributes:

nonessential = grid_external_power_172 - inverter_load_grid_169
essential = inverter_out_175 - (aux_power_166 - inverter_load_grid_169 )

and provide sensor etities for both values without doing 175-(166-169) as this is done inside SA already.

slipx06, 2023-05-10:
I could add the non-essential and essential sensor options to the card config and if they are set to 'internal' it uses the internal card logic to calculate otherwise the supplied sensor data

pawellygan, 2023-05-10:
> I could add the non-essential and essential sensor options to the card config and if they are set to 'internal' it uses the internal card logic to calculate otherwise the supplied sensor data

this!!!

slipx06, 2023-05-10:
non-essential and essential can now be specified through the card configuration. Setting these values to none uses the old logic. See v1.3.5

closed 2023-05-17 by slipx06