slipx06/sunsynk-power-flow-card #20: Auto convert W to kWh

shingsam, 2023-05-10:
Thank's for creating this beautiful power flow. Very useful for me who use old inverters.
Can u add auto convert W to kWh for calculating total daily?
Cause my entity use W, soo to long number on card.
Ex my value: 12.345 kWh why not 12.3 kWh.

Thanks.

slipx06, 2023-05-10:
Hi. Not sure I understand you correctly but if you are receiving wH and want to convert to kWh this is best handled outside of the card using a template sensor. e.g.

```
sensor:
  - platform: template
    sensors:
      wH_to_kWh:
	friendly_name: "Daily Energy kWh"
        value_template: "{{ (states('sensor.daily_Wh') | float /1000) | round(2) }}"
        unit_of_measurement: 'kWh' 
```

shingsam, 2023-05-13:
Oke, i know about it. But i mean need to add attribute unit W/kWh/MWh.

And, can u added battery current?

Thnks

slipx06, 2023-05-14:
The card expects the value in kWh so you need to modify your sensor accordingly.
Battery current has been added in v1.4.4

shingsam, 2023-05-15:
Okay thankyou...
I add entity on Battery current, but dnt show.
[image]
[image]

slipx06, 2023-05-15:
HI. Looks like you are still using an old version of the card. If you remove that sensor on the new version it will throw out a card error as it is required. Make sure you download v1.4.4. You can try clear your browser cache. If you using the companion app , you need to reset the frontend cache (IOS) or clear data cache (Android).

shingsam, 2023-05-15:
v1.4.4 now is show on Hac's.
Now is solved. Thanks

shingsam, 2023-05-15:
On lite mode can u added View like a battery information ?
[image]

amctear, 2023-05-15:
Yes, all working great.

slipx06, 2023-05-17:
I've changed the look and feel to move away from the grid layout.  Its updated in 1.4.5.  The lite view is meant to be simple. If you need the additional information for the load try the full card.

closed 2023-05-19 by slipx06