slipx06/sunsynk-power-flow-card #21: Minor Issue re Mobile connection to HA

amctear, 2023-05-14:
Hi, 
Absolutely wonderful job with this great power flow card, loving everthing about and promoted it in the Sunsynk FB forum. 
Having Dual Inverters in parallel, I've created a multitude of templates, if any one needs a copy. 

Runs perfectly via windows HA, after version 1.3 however, the IOS HA fell down.  Well it simply shows the entire card yaml only.  Thoughts ?

[image]

jacauc, 2023-05-15:
Kill the app on ios and try again

slipx06, 2023-05-15:
Thanks for the positive feedback. This happens when the app caches old resources. You can fix it by selecting the following from within the app

Settings - Companion App - Debugging - Reset frontend Cache

amctear, 2023-05-15:
Thanks, I had already reset the IOS Cache a few times before troubling you.  

Done some investigation on the IOS side earlier and found the issue. The IOS HA Card is faulting with an expectation of an extra entity battery_current_191:   It was asking for a sensor entry, so I simply added a battery_current_191: with a factitious sensor.

All working perfectly now and once again, thanks for the great card.

slipx06, 2023-05-15:
The additional battery current sensor was introduced in version 1.4.4.