slipx06/sunsynk-power-flow-card #22: SolarAssistant and wrong dot flow

pawellygan, 2023-05-19:
Today i was discharging my battery to the grid and the flow has the wrong direction. Dot is going towards battery instead towards the inverter

 Also I think value of wattage should be negative for discharging. SA is sending it as a  negative value but card seeing it as positive 
[image]

[image]

slipx06, 2023-05-19:
The card by design will always show a positive value for battery power regardless whether it is charging or discharging.  The flow direction is used to indicate this.

Have a look at the battery invert_power attribute.  You can set it to yes for your particular sensor to display the correct flow direction.

`invert_power: | Optional | no | Set to yes if your sensor provides a positive number for battery charge and negative number for battery discharge`

shingsam, 2023-05-20:
Hi @slipx06 i need more MPPT. 😆 
I have issue with no grid colour when status grid is off or unavailable.
And can u added dot animation on out image inverter when use full mode. Also added custom image inverter use format xml.
Thanks for the update. 👍

slipx06, 2023-05-20:
That looks like a large system and there is limited space. You will need to create some template sensors to combine some of your MPPT output i.e PV4 = PV4+PV5+PV6 etc
The grid status sensor grid_status_194 expect a 0/1 or off/on value. It does not appear to be working on your card so you should check the output of your sensor. The icon should change to a transmission line with line through it when grid is disconnected and will use the colour you specify in the config
I'll think about the extra animations. Dont want the card to be too busy.

shingsam, 2023-05-20:
> That looks like a large system and there is limited space. You will need to create some template sensors to combine some of your MPPT output i.e PV4 = PV4+PV5+PV6 etc The grid status sensor [...]

Yess i use it, with use 10 mppt.
Sensor value for grid is off, but icon not change.
For offgrid system they dnd need grid, and always with battery.
I always wait for the next update, thanks

slipx06, 2023-05-20:
There is something wrong with your card. configuration or sensor if the icon does not change. Create a template sensor that outputs a 1 or 0 or on or off and test.  

BTW it is case sensitive and needs to match exactly i.e. wont work with On, Off, ON, OFF, 'on', 'off' etc

shingsam, 2023-05-20:
> There is something wrong with your card. configuration or sensor if the icon does not change. Create a template sensor that outputs a 1 or 0 or on or off and test.
> 
> BTW it is case sensitive and needs to match exactly i.e. wont work with On, Off, ON, OFF, 'on', 'off' etc

Same dnd't work, maybe i dnt luck.
Can the next update add an option to hide grids like solar.
Thankyou

closed 2023-05-25 by slipx06