slipx06/sunsynk-power-flow-card #23: Feature Request - add wallboxes

creuter27, 2023-05-28:
Thank you for the great work - much appreciated!
I was wondering whether you could add up to 2 wall box chargers to the card?
In my setup, they are connected to the non-essential branch so the non-essential power value is the sum of the 2 wall box power values. So in my case, it would be perfect to replace the house symbol of the non-essential branch with 2 car symbols and 2 power values. The existing non-essential power value should stay so it can be configured to show the total non-essential power.
It is possible that some users may connect their wall box(es) to the load output but in my opinion that doesn't make much sense so adding them to the non-essential branch makes the most sense.
Also. I am using Tibber so the current cost per kWh grid power varies hourly and it would be great to be able to see the current price in the card as well. Could you add another value field for that somewhere in the grid branch?

slipx06, 2023-05-28:
Hi. Sounds like quite a unique use case. There is limited card space so you might want to track these on another card.  As I understand you will still have the total power in the non-essential sensor already?
Can you share the output of your Tibber sensor. I'm not familiar with it.

creuter27, 2023-05-29:
Hi slipx06,
I don't think that the use case is unique. A lot of PV system owners are driving electric vehicles as well and have a wall charger installed. Of course, I could display this information on another card but isn't that the case with almost all the data shown on this card? When one of the 2 chargers is in use then there is a very large amount of the total power in the system going to this endpoint. Definitely, something you want to see in the energy monitoring overview. Especially I want to see with a quick glance if the system is working correctly and using only excess power generated by the solar panels.
I know that the space is limited but I think if the house symbol for the non-essential load is replaced by 2 smaller car symbols (or up to 2 configurable icons??) then it should be possible to fit in.
The tibber sensor simply provides a number (eg 13 for 13 Cents per kWh). This value changes hourly.

creuter27, 2023-05-29:
In fact, I have considered forking the project and modifying the code myself but I am not familiar with creating SVG images and animations in javascript so I would have to go through quite some learning first. Are you using some kind of visual editor for the SVG code to simplify the testing?

Gorwanvfr, 2023-05-30:
> Hi. I dont use an SVG visual editor as most of the icons are SVG's positioned inside the main SVG container and the rest are just lines and rectangles etc. I usually just edit the SVG directly using a text editor and you can view the changes in any browser.
> Are you thinking something like this. Ignore the house. And then the two wall charges underneath?

Hello,     I found your power card last night. Absolutely fantastic. One question is it possible to add two bubbles out from the house to show car charging and water heating? This would be the icing on the cake. I love the fact that you have made a card that allows the DC battery to be shown so many apps and cards really only work for an AC coupled battery.  Thanks. Steve Sent from my iPhone

Gorwanvfr, 2023-05-30:
> I assume this is the same scenario as above i.e. non-essential loads?

Something like this

Gorwanvfr, 2023-05-30:
[image]

creuter27, 2023-05-30:
Hi, yes this is pretty much what I mean. I added the current Grid energy cost per kWh in the bottom right.
The daily totals would be nice to have as well.
More or less like this plus the 2 icons of a car or a wall charger or so.
That would make it perfect for me :-)
[image]

creuter27, 2023-05-30:
In order to make this more versatile it would probably be ideal to allow up to 2 non-essential loads that can be configured (name, icon, show total?). This would meet both requirements mentioned here and probably several others as well.
Also, (I don't need this but just just a thought) : Since The Aux power terminal can be configured as a smart load it would probably make sense to allow the same configuration options here as well (name, icon, show total?).

creuter27, 2023-06-01:
Hi slipx06,
I think it would also make sense to add the inverter heat sink temperature and the battery temperature. The inverter's Modbus output provides both.

Papsnoek, 2023-06-01:
Can you add another Bubble for Geyser output on the Essentials Side
Just kidding :) I work in software Dev, so I know how these request escalate into many different requests that are similar :)

My suggestion is to add these two bubbles as configuration settings under "load:" and "grid:" configuration as @creuter27 mentioned. Would mean then anyone can use it for any use and connect up any sensor they would like to show the power consumption of any device they need :) Only tricky part is "space" as now both grid and essentials could possible have two extra bubbles.

I will be adding some other features in my fork, so maybe I can help out with this when Im done.

slipx06, 2023-06-01:
:) I'm not a software developer so all contributions that improve the card/code are welcome. I'm going to have a look at adding the additional Loads to essential, non-essential and aux. Haven't quite figured out how to render mdi icons inside an SVG container so images may be generic for now but the idea is you can customize the label.  Space is definitely an issue

slipx06, 2023-06-01:
You can enable panel mode and setup a 1x1 grid. All the items are positioned relative to the SVG container, so everything will just scale though

slipx06, 2023-06-04:
@creuter27 I've added this to v1.6.2.  There are probably some visual improvements we can make with the icons in a later release

[image]

slipx06, 2023-06-04:
> [image]

Will look at how best to do this next. Not sure about the circles

closed 2023-06-08 by slipx06

slipx06, 2023-06-11:
@creuter27 I've included a sensor to track energy costs in v1.6.5

creuter27, 2023-06-12:
> @creuter27 I've included a sensor to track energy costs in v1.6.5

Awesome thank you so much!

toleabivol, 2023-12-29:
Hi, thank you for the awesome feature! 
One question, what am I doing wrong ?
I have a go-eCharger and would like to display the power that goes through it to the car and maybe (if possible) the daily energy it consumed. 
Now I get a 0 and under it the current power.

[image]

```yaml
square: true
type: grid
cards:
  - type: custom:sunsynk-power-flow-card
    cardstyle: compact
    show_solar: true
    battery:
      energy: 0
      shutdown_soc: 14
      show_daily: true
      auto_scale: true
    solar:
      show_daily: true
      mppts: 2
      pv1_name: House
      pv2_name: Garage
      dynamic_colour: true
      auto_scale: true
    load:
      show_daily: true
      dynamic_icon: true
      dynamic_colour: true
      auto_scale: true
      additional_loads: 1
      load1_name: go-eCharger
      show_aux: false
      load1_icon: mdi:ev-station
      load2_icon: ''
      aux_type: ''
      aux_loads: 0
      show_daily_aux: false
      show_absolute_aux: false
      aux_load1_icon: ''
    grid:
      show_daily_buy: true
      show_daily_sell: true
      show_nonessential: false
      auto_scale: true
      export_colour:
        - 25
        - 210
        - 71
      additional_loads: 0
    entities:
      use_timer_248: switch.ss_use_timer
      priority_load_243: switch.ss_priority_load
      inverter_voltage_154: sensor.ss_inverter_voltage
      load_frequency_192: sensor.ss_load_frequency
      inverter_current_164: sensor.ss_inverter_current
      inverter_power_175: sensor.ss_inverter_power
      inverter_status_59: sensor.ss_overall_state
      day_battery_charge_70: sensor.ss_day_battery_charge
      day_battery_discharge_71: sensor.ss_day_battery_discharge
      battery_voltage_183: sensor.ss_battery_voltage
      battery_soc_184: sensor.ss_battery_soc
      battery_power_190: sensor.ss_battery_power
      battery_current_191: sensor.ss_battery_current
      grid_power_169: sensor.ss_grid_power
      day_grid_import_76: sensor.ss_day_grid_import
      day_grid_export_77: sensor.ss_day_grid_export
      grid_ct_power_172: sensor.ss_grid_ct_power
      day_load_energy_84: sensor.ss_day_load_energy
      essential_power: none
      aux_power_166: sensor.sunsynk_aux_power
      day_pv_energy_108: sensor.ss_day_pv_energy
      pv1_power_186: sensor.ss_pv1_power
      pv2_power_187: sensor.ss_pv2_power
      pv1_voltage_109: sensor.ss_pv1_voltage
      pv1_current_110: sensor.ss_pv1_current
      pv2_current_112: sensor.deye_pv2_current
      pv2_voltage_111: sensor.deye_pv2_voltage
      essential_load1_extra: sensor.goecharger_go_echarger_p_all
      essential_load2_extra: sensor.goecharger_go_echarger_current_session_charged_energy
      aux_load1_extra: sensor.goecharger_go_echarger_p_all
      aux_load2_extra: sensor.goecharger_go_echarger_current_session_charged_energy
      aux_connected_status: sensor.goecharger_go_echarger_car_status
      energy_cost_buy: '0.32'
      energy_cost_sell: '0.082'
      grid_connected_status_194: binary_sensor.ss_grid_connected
      day_aux_energy: sensor.goecharger_go_echarger_current_session_charged_energy
    panel_mode: true
    show_battery: true
    show_grid: true
    large_font: true
    inverter:
      three_phase: true
      auto_scale: false
      modern: true
      autarky: 'no'
      colour:
        - 120
        - 120
        - 120
      model: sunsynk
columns: 1
```