slipx06/sunsynk-power-flow-card #28: Feature Request - add temperatures

creuter27, 2023-06-01:
Hi slipx06,
I think it would also make sense to add the inverter heat sink temperature and the battery temperature. The inverter's Modbus output provides both.

PS: I opened a new issue for this request just so it doesn't get lost. Should be quite easy to implement if you can find some space for it.

slipx06, 2023-06-01:
Added to v1.5.6 :)

Papsnoek, 2023-06-01:
Ah, you beat me to it, I was almost done testing my version :)
Time to go compare notes :P

Papsnoek, 2023-06-01:
Just FYI, this is how my implementation looks. I do however prefer yours :)
[image]

slipx06, 2023-06-01:
The thing is not everyone would want to display them.

Papsnoek, 2023-06-01:
yeah, I did make it optional, so if not displayed, then it would go back to original with only Power and Amps. I do like your design better through, much smaller, and I didnt add the battery temp like you did.

FYI: Thanks for creating this Card, its pretty darn amazing ;)

closed 2023-06-01 by slipx06