slipx06/sunsynk-power-flow-card #30: Feature Request - allow reverse grid_external_power_172

github-delusion, 2023-06-01:
Hello, could you allow set the reverse direction of energy flow from the grid?

Thank you very much

slipx06, 2023-06-01:
Hi. I'm not sure what you are requesting. The card already supports both grid import or export and will update the animation according to positive or negative power values i.e negative grid_external_power = grid export

github-delusion, 2023-06-01:
Yes, I think I have the possibility to invert the value of the sensor. I use this great card with GoodWe inverter . There I don't have a flow sensor on the inverter, but on the "watrouter" meter, the configuration looks like this:
grid_external_power_172: sensor.active_power.
Since it is a sensor on the meter the flow is logically reversed. It's the same with batteries, but there you have the option to turn the current in the script.... I mean this part:

 let battery_power = "";
    if (config.battery.invert_power === 'yes'){
        battery_power = (parseInt(stateObj13.state) * -1);

And it would help me if the item grid_external_power_172 could be rotated in the same way .....

slipx06, 2023-06-02:
Added to v1.5.7 invert_grid: yes/no

github-delusion, 2023-06-02:
Hi, Thank you very much, now this it works correctly for me!

Big thanks