slipx06/sunsynk-power-flow-card #31: The "inverter out 175" item popup does not work.

github-delusion, 2023-06-02:
Hello,
I use the sensor "sensor.house_consumption" to populate the "inverter_out_175" item. After clicking on the image or the value, the window does not appear 
[image]
 . But the sensor "sensor.house_consumption" has a pop-up window with a graph 
[image]
.
So I think it's a shame it doesn't work.

It would be lovely if you you could look at it.

Thank you very much

slipx06, 2023-06-02:
Hi. That value comes from the essential_power sensor. If you don't supply a sensor the card uses internal calculation which is why it is displayed but then you cannot click on it as there is no historical data. Update your card configuration for it to work.

essential_power: sensor.house_consumption

github-delusion, 2023-06-02:
Hello,

Yes, you are right again. That is very well.
It was enough to add the sensor you wrote to the configuration "essential_power: sensor.house_consumption" nd popup window is function.
You are great and I thank you very much.

FYI my current configuration:

type: custom:sunsynk-power-flow-card
cardstyle: lite
show_solar: 'yes'
battery:
  energy: 24000
  shutdown_soc: 10
  show_daily: 'yes'
  invert_power: 'no'
solar:
  show_daily: 'yes'
  mppts: two
load:
  show_daily: 'yes'
inverter:
  modern: 'yes'
  colour: green
  autarky: 'yes'
grid:
  invert_grid: 'yes'
  show_daily_buy: 'yes'
  show_daily_sell: 'yes'
  show_nonessential: 'no'
entities:
  essential_power: sensor.house_consumption
  use_timer_248: xes
  priority_load_243: 'no'
  batdischargeday_71: sensor.today_battery_discharge
  batchargeday_70: sensor.today_battery_charge
  loadday_84: sensor.today_load
  inverter_grid_voltage_154: sensor.on_grid_l1_voltage
  inverter_load_freq_192: sensor.meter_frequency
  inverter_out_164: sensor.on_grid_l1_current
  inverter_out_175: sensor.house_consumption
  grid_status_194: binary_sensor.grid_connected_status
  inverter_status_59: sensor.overall_state
  battery_voltage_183: sensor.battery_voltage
  battery_soc_184: sensor.battery_state_of_charge
  battery_out_190: sensor.battery_power
  battery_current_191: sensor.battery_current
  inverter_load_grid_169: sensor.grid_power
  grid_buy_day_76: sensor.homekit_none_import
  grid_sell_day_77: sensor.homekit_none_export
  grid_sell_day_sensor.grid_import_day_buy77: sensor.grid_export_day_sell
  grid_external_power_172: sensor.active_power
  aux_power_166: sensor.aux_output_power
  solarday_108: sensor.today_s_pv_generation
  pv1_power_186: sensor.pv1_power
  pv2_power_187: sensor.pv2_power
  pv1_v_109: sensor.pv1_voltage
  pv1_i_110: sensor.pv1_current
  pv2_v_111: sensor.pv2_voltage
  pv2_i_112: sensor.pv2_current

closed 2023-06-02 by github-delusion