slipx06/sunsynk-power-flow-card #32: Total energy import/export doesn't work for me.

github-delusion, 2023-06-02:
Hello,

the daily sum of exported and imported energy does not work for me
[image]

. Unfortunately, I couldn't find the right sensor to solve this value.
But it has to work somehow, because the "power-flow-card-plus" card can do it 
[image]
, I just don't know how to display this value on your card :-(

My config:

type: custom:sunsynk-power-flow-card
cardstyle: lite
show_solar: 'yes'
battery:
  energy: 24000
  shutdown_soc: 10
  show_daily: 'yes'
  invert_power: 'no'
solar:
  show_daily: 'yes'
  mppts: two
load:
  show_daily: 'yes'
inverter:
  modern: 'yes'
  colour: green
  autarky: 'yes'
grid:
  invert_grid: 'yes'
  show_daily_buy: 'yes'
  show_daily_sell: 'yes'
  show_nonessential: 'yes'
entities:
  use_timer_248: xes
  priority_load_243: 'no'
  batdischargeday_71: sensor.today_battery_discharge
  batchargeday_70: sensor.today_battery_charge
  loadday_84: sensor.today_load
  inverter_grid_voltage_154: sensor.on_grid_l1_voltage
  inverter_load_freq_192: sensor.meter_frequency
  inverter_out_164: sensor.on_grid_l1_current
  inverter_out_175: sensor.house_consumption
  grid_status_194: binary_sensor.grid_connected_status
  inverter_status_59: sensor.overall_state
  battery_voltage_183: sensor.battery_voltage
  battery_soc_184: sensor.battery_state_of_charge
  battery_out_190: sensor.battery_power
  battery_current_191: sensor.battery_current
  inverter_load_grid_169: sensor.grid_power
  grid_buy_day_76: sensor.grid_import_day_buy
  grid_sell_day_77: sensor.grid_export_day_sell
  grid_sell_day_sensor.grid_import_day_buy77: sensor.grid_export_day_sell
  grid_external_power_172: sensor.active_power
  aux_power_166: sensor.aux_output_power
  solarday_108: sensor.today_s_pv_generation
  pv1_power_186: sensor.pv1_power
  pv2_power_187: sensor.pv2_power
  pv1_v_109: sensor.pv1_voltage
  pv1_i_110: sensor.pv1_current
  pv2_v_111: sensor.pv2_voltage
  pv2_i_112: sensor.pv2_current

Thank you very much

slipx06, 2023-06-02:
What do you see under developer tools - states? You should be able to find the correct sensors there. If you are not reading these values from the inverter you can setup an Energy meter within home assistant to populate for you

github-delusion, 2023-06-02:
Hello,

Yes you are right. I found sensors:
sensor.homekit_none_export
sensor.homekit_none_import

and now it looks functional.

Many thanks.

[image]

closed 2023-06-02 by github-delusion