slipx06/sunsynk-power-flow-card #37: Rename PV1...4 ?

Snux, 2023-06-06:
I'm using this card for displaying information on my Growatt system, I found equivalent sensors for almost everything.  I prefer this card to others I've found as it's the only one showing the multiple PV strings.

A really small and not very important request, but could we have the option to change the title for PV1 through to PV4?  I would like to rename to PV1 to "East" and PV2 to "West" so my family know which is which.

Also, if a sensor is set to "none" could it be removed from the display?  I have a few that I can't get, so they just display with zero.

I know this isn't quite the use case this was built for so no problem if you don't want to do either of these :)

Mark

[image]

slipx06, 2023-06-06:
I'll look into adding both of those requests. Thanks for the feedback :)

slipx06, 2023-06-06:
@Snux give 1.6.4 a go

MikeRobertson, 2023-06-06:
Perfect! Thanks very much!

Snux, 2023-06-07:
Awesome, works just right!  I'll close this one

closed 2023-06-07 by Snux