slipx06/sunsynk-power-flow-card #38: Animation query

JayMcG, 2023-06-09:
Hey. I've configured the card but using data from Solar Assistant. The data is real-time, meaning that I can see value changes every few seconds. I've noted that the dot animations reset when the wattage data changes making the card look like it's glitching. I'm not sure if this is something that can easily be resolved. Thanks in advance for the cool card!

slipx06, 2023-06-09:
Hi. Can you share the history for one of your power sensors where you notice the problem.  This could happen if the values are changing significantly on each refresh as it recalculates the animation speed. I don't notice if when the sensor data changes are small. I also have my refresh set to 10sec. As an interim fix you can disable the dynamic animations by removing the   animation_speed: and  max_power: attributes and it will use fixed speeds. Alternatively you can increase the max_power values to make it less sensitive to power fluctuations.

JayMcG, 2023-06-09:
Hi. There can be some big fluctuations but I see it seems to do it even on smaller changes (e.g 5w). My inverter is a 5000w system. I tried increasing the max_power to 10000w and speed to 5 but seems to have the same result.

slipx06, 2023-06-11:
It's a limitation of the current animation code. It does not handle this use case well

closed 2023-11-13 by slipx06

slipx06, 2023-11-13:
@JayMcG can you test if this fixes your issue.

JayMcG, 2023-11-23:
@slipx06 looks like it, nice one!

PR #132 (2023-11-13): Feature/animation rework
Includes changes from https://github.com/slipx06/sunsynk-power-flow-card/pull/131

## Description

## Related issues
Resolves #38 

## Motivation and Context
Smoothes out animation transition.