slipx06/sunsynk-power-flow-card #40: Grid power sensor

zlakes01, 2023-06-18:
Hi. I have Solax inverter with two grid sensors one for import and one for export. Is it posible to use those two sensors for grid_external_power_172  entity?

slipx06, 2023-06-18:
Hi
You should create a template sensor that combines these two values. Assuming both sensors report a positive number for grid import and grid export you can use the template below as the card expects a negative value for grid export. Replace the sensor names. If your grid export sensor is already negative then just swap those two sensors around in the formula below

```
- platform: template
  sensors:
    grid_power:
      unit_of_measurement: "W"
      value_template: "{{ states('sensor.grid_import')|float - states('sensor.grid_export')|float }}"
```

zlakes01, 2023-06-19:
Thanks a lot. I have done a new sensor and will test it.

closed 2023-06-22 by slipx06

MarekMazurek, 2024-08-07:
Hello. I followed the advice above and modified the code according to my entities. The entity works. Unfortunately, when I indicate it in the card as grid_power, the card does not respond. What could I be doing wrong?