slipx06/sunsynk-power-flow-card #41: Making 109/110/11/112 Optional

guybw, 2023-06-19:
Hi slipx06,
I'm loving your project! 
I'm not using Synsynk I'm using a different inverter but your card is just brillant!
The only thing is right now, my inverter doesn't show amps or volts (yet) 
Could you help make the following optional at all or am I missing something?  
pv1_v_109
pv1_i_110
pv2_v_111
pv2_i_112

Thanks!

slipx06, 2023-06-19:
Hi. Thank you.
Just set them to none and they will be hidden

```
  pv1_v_109: none
  pv1_i_110: none
  pv2_v_111: none
  pv2_i_112: none
```

guybw, 2023-06-19:
Thanks, I so totally missed that! works amazingly well.Thanks for the great work!

closed 2023-06-19 by guybw

slipx06, 2023-06-19:
If you update to the latest release (v1.7.4) you can delete them from your card config and it will no longer complain about missing sensors