slipx06/sunsynk-power-flow-card #42: Autarky/Ratio issues

elRadix, 2023-06-21:
Hi @slipx06
Using the latest avaliable version of your card still notice some issues
The value can't be negative and can't exceed 100%

Can you please check?

some information:
https://www.npro.energy/main/en/help/self-sufficiency-rate

[image]

slipx06, 2023-06-21:
Are you sure you are running the latest version and cleared your cache. Version 1.7.5 uses the formula below

```
    let production_p = parseInt(totalsolar) + parseInt(`${battery_power > 0 ? battery_power : 0}`) + parseInt(`${aux_power < 0 ? aux_power * -1 : 0}`);
    let consumption_p = parseInt(essential) + parseInt(nonessential) + parseInt(`${aux_power > 0 ? aux_power : 0}`) + parseInt(`${battery_power < 0 ? battery_power * -1 : 0}`);
    let Autarkyp = consumption_p != 0 ? Math.min(Math.round((production_p * 100) / consumption_p), 100) : 0;
    let Ratiop = production_p != 0 ? Math.min(Math.round((consumption_p * 100) / production_p), 100) : 0;
```

Do you provide sensors for your essential and non-essential or do you let the card calculate this?

According to your screenshot 

```
production = 9635 + 0 + 0
consumption = 752 + nonessential + 0 + 0

Autarky = 752/9635 = 8%
Ratio = 9635/752 = 100%
```

slipx06, 2023-06-21:
[image]

slipx06, 2023-06-24:
@elRadix did you get this working?

slipx06, 2023-07-02:
Assume this is working now so closing.

closed 2023-07-02 by slipx06