slipx06/sunsynk-power-flow-card #46: Load 1 and 2 icons

Gorwanvfr, 2023-06-28:
Would it be possible to add the facility to add icons to load 1 and 2 on essential loads? The same as you have for nonessential loads.

slipx06, 2023-06-28:
It's possible but I cant see how these would fit on the card given the limited space

slipx06, 2023-06-30:
Added to v1.8.0

wacalitz, 2023-06-30:
Thanks, can a stove icon be added as well in a future update.  
Also I notice the icons only show in the full layout not the lite.  Any possibility to add this to the lite layout 
Perhaps here:
[image]
Thanks

slipx06, 2023-06-30:
Yes. I'll add it.  I didn't have time to work on the lite card. Will do that this weekend.

slipx06, 2023-07-01:
Added to v1.8.1
[image]

slipx06, 2023-07-01:
You can now use mdi icons in 1.8.2 😄 

[image]

Gorwanvfr, 2023-07-02:
[image]

Gorwanvfr, 2023-07-02:
One small issue with the mdi icons. When the card is in panel mode the mdi icons are in the middle of the card. 
[image]

When the card is not in panel mode then the icons are offset to the left. 
[image]

slipx06, 2023-07-02:
That's strange, I cant reproduce that.  Does it happen on all your devices? What browser are you using?

slipx06, 2023-07-02:
Seems to be a known issue with Safari.

Gorwanvfr, 2023-07-02:
Fixed positions looks to be working. 
[image]

closed 2023-07-03 by slipx06