slipx06/sunsynk-power-flow-card #47: MDI icons wrong size and position in Safari

slipx06, 2023-07-02:
There is an issue with the way Safari renders the foreignObject element which I use to display the mdi icons. This can affects their position and size. I don't know how to fix this so for now the work around is to use another browser or use the built in icon presets.

Pho3niX90, 2023-11-12:
@slipx06 is this only happening for safari? Any other browsers like brave, firefox etc? I don't own apple products so cannot reproduce

slipx06, 2023-11-12:
@Pho3niX90 only seems to be a problem with Safari

https://stackoverflow.com/questions/51313873/svg-foreignobject-not-working-properly-on-safari
https://discussions.apple.com/thread/254680843

Pho3niX90, 2023-11-12:
@slipx06 it seems this might be related to safari 16.3?
https://discussions.apple.com/thread/254680843

Pho3niX90, 2023-11-12:
I will see if I can maybe emulate safari

slipx06, 2023-11-14:
Thanks. It only becomes a problem if you are using mdi icons in the card. In other words load1_icon: mdi: example etc

I think there must be a better way of rendering the icon inside the SVG. I'll also take another look