slipx06/sunsynk-power-flow-card #48: Installation

Guenni75, 2023-07-15:
Hello.

I can't find anything about it in HACS??

slipx06, 2023-07-15:
Hi

You need to add to HACS as a Custom Repo. 

[Click to manually add to HACS](https://my.home-assistant.io/redirect/hacs_repository/?repository=sunsynk-power-flow-card&category=plugin&owner=slipx06)

milsky99, 2023-07-15:
Brilliant work thanks so much !!!

So I have been following the instructions of Solar Integrations on YouTube, the issue is he is linking dropbox files that appear to have been updated. I have downloaded the card code from this git hub and have it up and running now but am missing some key sensor data like solar watts and more.

I suspect that I need to update the Config.yaml file for some of the updated sensor names.... yet I cannot find this code on the GitHub, Im not massively technical but am sure im missing something basic

Pls help !

slipx06, 2023-07-16:
Hi. The card displays data based on the sensors you provide. Depending on how you are collecting the data from the inverter sensor names can change. I have shared my ESPHome config that will work with the the default card values. You can download it [here](https://github.com/slipx06/Sunsynk-Home-Assistant-Dash). If you are using other methods i.e. Solarman etc your best place is to visit the https://powerforum.co.za/topic/17970-home-assistant-sunsynk-power-flow-card-and-dashboard/ forums and ask. I have tried to map some of the more common integrations in the [WIKI](https://github.com/slipx06/sunsynk-power-flow-card/wiki)

milsky99, 2023-07-16:
Heya Slip

First off thanks again for putting all of this effort in, truly appreciated. So I did nav to your link but am not using ESPHome config but rather the one from Solar Integrations, I have managed to map many of the sensor but not all of them. Will continue to try find solutions here

Thanks again

closed 2023-07-21 by slipx06