slipx06/sunsynk-power-flow-card #50: PowerFlow Card viewing from Desktop vs Mobile

HarrisHome, 2023-07-27:
After the last update I am able to view the PowerFlow Card on my Desktop via Chrome happily in all it's glory (Animations, Colour, Live Data etc). However when viewing the same dashboard via the HomeAssistant App on my mobile, it only shows the code. All the entities are correctly configured in the code, but I am getting the following on the top "Please include the day_battery_charge_70 and day_battery_discharge_71 attributes and entity IDs" I scroll down to that section of code and all entities there are correct. Am I missing something? Screenshots attached.
[image]
[image]
[image]

slipx06, 2023-07-27:
I assume you recently updated. The home assistant companion app is caching old data.  You can fix it be clearing your app cache

Ikitse, 2023-07-28:
Same issue here. Both on desktop and the app. And no changes were made. Tried clearing data on app, but it doesn't work.

HarrisHome, 2023-07-28:
Correct, after the update the dashboard in the Mobile App went wonky.
Looked all over within the app (Android Mobile) for a clear cache setting but couldn't find anything.
The only cache I could find was in the App's list, data used etc. Cleared cache there but still the same.
Updated to version 1.9.2 which just popped up but still the same... Something broke...

HarrisHome, 2023-07-28:
Well now after updating to version 1.9.2 my local desktop dashboard is also gone belly up.
What do they normally say - "Don't fix what aint broke".

slipx06, 2023-07-28:
There was a breaking change in v1.8.6 Did you update your entity names?

https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v1.8.6

<html>
<body>
<p style="box-sizing: border-box; margin-top: 0px !important; margin-bottom: 16px; color: rgb(31, 35, 40); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 16px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><strong style="box-sizing: border-box; font-weight: var(--base-text-weight-semibold, 600);">BREAKING CHANGE</strong><br style="box-sizing: border-box;">The following entities have been renamed. Please update your card configuration with the new names:</p>

Old Name | New Name
-- | --
batchargeday_70: | day_battery_charge_70:
batdischargeday_71: | day_battery_discharge_71:
loadday_84: | day_load_energy_84:
grid_buy_day_76: | day_grid_import_76:
grid_sell_day_77: | day_grid_export_77:
solarday_108: | day_pv_energy_108:
inverter_grid_voltage_154: | inverter_voltage_154:
inverter_load_freq_192: | load_frequency_192:
inverter_out_164: | inverter_current_164:
inverter_out_175: | inverter_power_175:
inverter_load_grid_169: | grid_power_169:
battery_out_190: | battery_power_190:
grid_external_power_172: | grid_ct_power_172:
pv1_v_109: | pv1_voltage_109:
pv1_i_110: | pv1_current_110:
pv2_v_111: | pv2_voltage_111:
pv2_i_112: | pv2_current_112:
pv3_v_113: | pv3_voltage_113:
pv3_i_114: | pv3_current_114:
pv4_v_115: | pv4_voltage_115:
pv4_i_116: | pv4_current_116:
grid_status_194: | grid_connected_status_194:
battery_temp: | battery_temp_182:
inverter_ac_temp: | radiator_temp_91:
inverter_dc_temp: | dc_transformer_temp_90:

</body>
</html>

Ikitse, 2023-07-28:
Apologies missed this update. But this resolves everything! :-)

HarrisHome, 2023-07-28:
Same with me. Seemed to have missed those breaking changes.
Renamed entities and sorted.
Thanks for the quick response and help, MUCH APPRECIATED!

closed 2023-07-28 by slipx06