slipx06/sunsynk-power-flow-card #51: [FEATURE REQUEST] Clickable Total Solar entity

NemesisXB, 2023-07-31:
It would be great if the total solar entity is clickable. If a user does not have a combined entity the automatic calculation can be used. Maybe a user can select "auto" as the "entity" for automatic calculation (which will then not be clickable)
For example if you have a combined entity: combined_solar_power: sensor.deye_combined_solar
and if a user does not have a combined entity: combined_solar_power: auto
[image]

Quintinvw, 2023-07-31:
That would be a great addition.

slipx06, 2023-07-31:
Good idea. Added to v1.9.4

NemesisXB, 2023-07-31:
Wow. That was Quick. Seems to work great. Thanks

closed 2023-08-01 by slipx06