slipx06/sunsynk-power-flow-card #52: AUX_Load query? Feature?

djagerif, 2023-08-08:
I was wondering if it's possible to add aux_load1 and aux_load2 with their accompanying name etc. for metered loads on the AUX output similar to essential/non_essential loads?

slipx06, 2023-08-10:
Sure. Let me see what I can do

slipx06, 2023-08-10:
I've added this to v1.9.6

[image]

djagerif, 2023-08-10:
Looks great. My icons are just a bit off the side. How do I make it fit the standard HA defined widths?
[image]

slipx06, 2023-08-10:
Strange. What view type is that? Mine are not cut off? Have you changed the card height?
 
[image]

[image]

djagerif, 2023-08-10:
Ah, yes. Setting it back to standard height now shows them 'exactly' on the edge. My thought was with a height adjustment I might stretch the icon spacing a bit but that was not to be. Maybe a bit of a massage with the edges would be great but all good thanks.
[image]

slipx06, 2023-08-10:
Let me see if I can improve the layout. I might shift the daily load over and move the icons to the left. There is not much space to play with.

djagerif, 2023-08-10:
Please also check if aux_load2 is not defined then it should only add one entry to the UI.

slipx06, 2023-08-10:
At the moment it’s a yes/no for the two additional loads. If there is a requirement for one/two/no loads I can add that.

djagerif, 2023-08-11:
Yes, please. Similar to the 'additional_loads:' for essential.

slipx06, 2023-08-11:
OK give v1.9.7 a go

djagerif, 2023-08-11:
Perfect, thanks.