slipx06/sunsynk-power-flow-card #55: Request total_solar: sensor.lux_solar_output_live has the option to view the actual sensor graph

CaptainMidnight, 2023-08-13:
Hi

Thanks for such a great dashboard that I've configured for use with my Lux Hybrid inverter using @guybw LuxPython_DEV integration.

Looking at your dashboard options, using the total_solar: sensor.lux_solar_output_live, instead of the default sum of PV1+PV2 etc, is there any chance that this item on the dashboard could link to the related sensor graph?

i.e. Pv1 and PV2 are links to the individual solar input sensors and when clicked load the related sensor graphs, using a sensor with total_solar, shows the total solar input value and can link to the associated graph but in your dashboard doesn't offer that option.

In short - if you use a sensor as apposed to adding values together, could we also have the availability of opening the sensor graph please?

Phil

CaptainMidnight, 2023-08-13:
From reviewing some of your examples, the last one showing full configuration details: -

pv_total: sensor.sunsynk_totalsolar

but if I use this as follows: -

pv_toatl: sensor.lux_solar_output_live

it still doesn't give access to the functionality of being able to click and open the senor graph?

slipx06, 2023-08-13:
Hi. It should already do that. Are you sure that sensor.lux_solar_output_live is setup correctly in HA. It needs to have an associated unit of measurement for the historical graph to work.

[image]

CaptainMidnight, 2023-08-13:
Thank you @slipx06, after reading/watching your reply, on checking my own configuration I'd realised that I simply hand't saved the configuration change - so it was still just doing the automatic sum.

sorry, me bad it is working as demonstrated above.... and there's so many questions I could ask around various elements on that dashboard in your reply, but I better do a bit more 'proper' settings testing first.

Really appreciated the quick reply though!

CaptainMidnight, 2023-08-14:
Hi @slipx06 my friend, sorry I'm back with the same issue but on a different element of the card - this time, it's: -

  essential_load1: sensor.value_tapo_total
  essential_load2: sensor.phoenix_plug_06_current_power

Load1 I expect to not be clickable to view a graph as it's my own combination sensor that just sums up 5x other graph capable sensors ..... no issues there as I didn't expect it to work.

Load2 though can be programmed with any graphable sensor like the tapo plug as above or replaved with pv1 solar output sensor etc BUT it's still not allowing the option to display the related sensor graph, the same initial scenario I started with above.

I must be doing something wrong, or editing incorrectly but for the life of me I can't get these small essential boxes to allow for the sensor graph to be accessed?

CaptainMidnight, 2023-08-14:
In essence are 

  essential_load1: 
  essential_load2: 

in your card design also setup to be used with graphable sensors?

CaptainMidnight, 2023-08-14:
Much appreciated my friend :o)

slipx06, 2023-08-18:
Added to v2.0.0

CaptainMidnight, 2023-08-18:
Thank you, those load graphs are working great !

closed 2023-08-18 by CaptainMidnight