slipx06/sunsynk-power-flow-card #60: Lux Status Code Missing?

bazcurtis, 2023-08-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

I can see the dot next to the Inverter, but I can't see the code.

I would also like to thank you for support Lux Inverters.

[image]

### Expected behaviour

I assume a code should be shown somewhere

### Card Version

2.0.0

### Home Assistant Version

2023.8.2

### Configuration

```shell
type: custom:sunsynk-power-flow-card
cardstyle: lite
show_solar: true
inverter:
  model: lux
  modern: true
battery:
  energy: 6400
  shutdown_soc: 10
  show_daily: true
  invert_power: true
  colour: green
solar:
  show_daily: true
  mppts: 2
  pv1_name: Back
  pv2_name: Front
load:
  show_daily: true
grid:
  show_daily_buy: true
  show_daily_sell: true
  show_nonessential: false
  invert_grid: true
entities:
  day_pv_energy_108: sensor.lux_solar_output_daily
  inverter_voltage_154: sensor.lux_grid_voltage_live
  load_frequency_192: sensor.lux_grid_frequency_live
  inverter_current_164: none
  inverter_power_175: sensor.lux_battery_flow_live
  grid_connected_status_194: binary_sensor.binary_on_grid
  inverter_status_59: sensor.lux_status
  day_battery_charge_70: sensor.lux_battery_charge_daily
  day_battery_discharge_71: sensor.lux_battery_discharge_daily
  battery_voltage_183: sensor.lux_battery_voltage_live
  battery_soc_184: sensor.lux_battery
  battery_power_190: sensor.lux_battery_flow_live
  battery_current_191: none
  grid_power_169: sensor.grid_power
  day_grid_import_76: sensor.lux_power_from_grid_daily
  day_grid_export_77: sensor.lux_power_to_grid_daily
  grid_ct_power_172: sensor.lux_grid_flow_live
  day_load_energy_84: sensor.lux_home_consumption_daily
  essential_power: sensor.lux_home_consumption_live
  nonessential_power: none
  aux_power_166: none
  pv_total: sensor.lux_solar_output_live
  pv1_power_186: sensor.lux_solar_output_array_1_live
  pv2_power_187: sensor.lux_solar_output_array_2_live
  pv1_voltage_109: none
  pv1_current_110: none
  pv2_voltage_111: none
  pv2_current_112: none
  inverter_ac_temp: sensor.lux_radiator_1_temperature_live
  inverter_dc_temp: sensor.lux_radiator_2_temperature_live
  remaining_solar: sensor.solcast_forecast_remaining_today
  energy_cost: sensor.octopus_agile_current_rate
view_layout:
  grid-area: r4
```

CaptainMidnight, 2023-08-18:
@bazcurtis if you switch to 'full' instead of 'lite' you'll see it's working, but in the 'lite' view maybe just gives the dot based on space available etc .... : -

[image]

slipx06, 2023-08-18:
Yes the lite view only shows the coloured dot.

You can see the list of codes below

https://github.com/slipx06/sunsynk-power-flow-card/blob/833e2b37b323e9c7a58c79e2ca45cf0f2d89e9e8/src/const.ts#L8-L36

marksie1988, 2023-08-18:
There are possibly missing codes, I could only add a limited number as I don't have a full list from anywhere.

CaptainMidnight, 2023-08-18:
..... from [sensor.py](https://github.com/guybw/LuxPython_DEV/blob/master/custom_components/luxpower/sensor.py)

    def push_update(self, event):
        _LOGGER.debug(f"Sensor: register event received Bank: {self._bank} Attrib: {self._device_attribute} Name: {self._attr_name}")  # fmt: skip
        self._data = event.data.get("data", {})
        state_text = ""
        status = int(self._data.get(self._device_attribute, 0.0))
        # fmt: off
        if status == 0:
            state_text = "Standby"
        elif status == 1:
            state_text = "Error"
        elif status == 2:
            state_text = "Inverting"
        elif status == 4:
            state_text = "Solar > Load - Surplus > Grid"
        elif status == 5:
            state_text = "Float"
        elif status == 7:
            state_text = "Charger Off"
        elif status == 8:
            state_text = "Supporting"
        elif status == 9:
            state_text = "Selling"
        elif status == 10:
            state_text = "Pass Through"
        elif status == 11:
            state_text = "Offsetting"
        elif status == 12:
            state_text = "Solar > Battery Charging"
        elif status == 16:
            state_text = "Battery Discharging > LOAD - Surplus > Grid"
        elif status == 17:
            state_text = "Temperature Over Range"
        elif status == 20:
            state_text = "Solar + Battery Discharging > LOAD - Surplus > Grid"
        elif status == 32:
            state_text = "AC Battery Charging"
        elif status == 40:
            state_text = "Solar + Grid > Battery Charging"
        elif status == 64:
            state_text = "No Grid : Battery > EPS"
        elif status == 136:
            state_text = "No Grid : Solar > EPS - Surplus > Battery Charging"
        elif status == 192:
            state_text = "No Grid : Solar + Battery Discharging > EPS"
        else:
            state_text = "Unknown"
        self._attr_native_value = f"{state_text}"
        # fmt: on

marksie1988, 2023-08-19:
> ..... from [sensor.py](https://github.com/guybw/LuxPython_DEV/blob/master/custom_components/luxpower/sensor.py)
>
> ```
> def push_update(self, event):
>     _LOGGER.debug(f"Sensor: register event received [...]

That's excellent thanks, I'll get them updated as soon as I get chance. Possibly Sunday unless slipx can get to it sooner.

bazcurtis, 2023-08-19:
Thanks very much. It is interesting there are so many codes. I have only seen these.

20 - Solar + Battery Discharging > LOAD - Surplus > Grid (On Grid as well)
12 - Solar > Battery Charging
4 - Solar > Load - Surplus > Grid
0 - Standby
32 - AC Charging
16 - Battery Discharging > LOAD - Surplus

PR #61 (2023-08-19): feat: add additional lux status codes

## Description

A small patch to add additional status codes for Lux inverters

## Related issues

closes #60 

## Motivation and Context

Allows more cover for status codes in the Lux inverter. 

## How has this been tested

I have tested this locally but cant fully test the state codes until they occur

## Type of change

Please delete options that are not relevant.

- [ ] Bug fix (non-breaking change which fixes an issue)
- [x] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing fun [... cut]
closed 2023-08-19 by slipx06