slipx06/sunsynk-power-flow-card #63: Show the Octopus Export and Import rate depending on direction

bazcurtis, 2023-08-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Shows the import rate

### Expected behaviour

Currently works as designed

### Possible Solutions

Show the Octopus Export and Import rate depending on direction. Change colour or add an arrow and show the current price depending on if you are importing or exporting

### Mode

Both

### Context / Reason

It shows importing all the time, even if you are exporting. I thought it might be nice to the price on what is happening at the time.

This is the best energy card out there for Home Assistant in my view.

slipx06, 2023-08-21:
Hi. I have no way of testing this so will need a bit more information. 
The card shows import or export based on the animated power flow and the power value (+-). 
Does Octopus provide separate sensors when importing or exporting?

marksie1988, 2023-08-24:
> Hi. I have no way of testing this so will need a bit more information. The card shows import or export based on the animated power flow and the power value (+-). Does Octopus provide separate sensors when importing or exporting?

The octopus integration provides the price for import and price for export. 

assuming I understand correctly, we could update the `energy_cost` with the following: 

```
export_price
import_price
```

Then depending on the  flow of the electricity we can pick which of these options to use?

bazcurtis, 2023-08-25:
I am using this [Integration](https://github.com/BottlecapDave/HomeAssistant-OctopusEnergy/)

it has these two sensors.

sensor.agile_current_import_rate
sensor.agile_current_export_rate

I am happy to do any testing if you decide to implement this

[image]

slipx06, 2023-08-25:
Ok I understand. It can be implemented as per @marksie1988 suggestion above

slipx06, 2023-08-25:
I've added this to https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/2.2.0

<html>
<body>

Attribute | Requirement | Default | Description
-- | -- | -- | --
energy_cost_buy: | Optional |   | Sensor that provides current buy energy cost per kWh
energy_cost_sell: | Optional |   | Sensor that provides current sell energy cost per kWh

</body>
</html>

The card will display the correct cost based on grid power

bazcurtis, 2023-08-26:
Thanks for adding this. The sun has just gone in and I am using the battery 😄. I will check it.

bazcurtis, 2023-08-27:
Working fantastic. Thank you very much.

closed 2023-08-27 by bazcurtis