slipx06/sunsynk-power-flow-card #64: Show Daily not working for Solar, Grid and Battery after latest update

johannwilken, 2023-08-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

After updating to the latest version the show daily for Solar, Gid and Battery is not showing anymore.

### Expected behaviour

It was showing before update

### Card Version

2.1.1

### Home Assistant Version

8.3

### Configuration

```shell
type: custom:sunsynk-power-flow-card
cardstyle: lite
show_solar: 'yes'
card_height: 320px
inverter:
  modern: 'yes'
battery:
  energy: 10240
  shutdown_soc: 20
  show_daily: 'yes'
  invert_power: 'yes'
  animation_speed: '5'
  max_power: '5000'
  full_capacity: '100'
  empty_capacity: '20'
solar:
  show_daily: 'yes'
  mppts: 2
  animation_speed: '5'
  max_power: '6000'
load:
  show_daily: 'yes'
  animation_speed: '5'
  max_power: '6000'
grid:
  show_daily_buy: 'yes'
  show_daily_sell: 'no'
  show_nonessential: 'no'
  animation_speed: '5'
  max_power: '10000'
entities:
  use_timer_248: switch.use_timer
  inverter_power_175: sensor.house_power
  day_battery_charge_70: sensor.battery_energy_in
  day_battery_discharge_71: sensor.battery_energy_out
  battery_voltage_183: sensor.battery_voltage
  battery_soc_184: sensor.battery_state_of_charge
  battery_power_190: sensor.battery_power
  battery_current_191: sensor.battery_current
  grid_power_169: sensor.grid_power
  day_grid_import_76: sensor.grid_todays_energy
  day_grid_export_77: sensor.grid_energy_out
  grid_ct_power_172: sensor.grid_power_ct
  day_load_energy_84: sensor.house_todays_energy
  essential_power: sensor.house_power
  nonessential_power: none
  day_pv_energy_108: sensor.pv_todays_energy
  pv1_power_186: sensor.pv_1_power
  pv2_power_187: sensor.pv_2_power
  pv1_voltage_109: sensor.pv_1_voltage
  pv1_current_110: sensor.pv_1_current
  pv2_voltage_111: sensor.pv_2_voltage
  pv2_current_112: sensor.pv_2_current
  radiator_temp_91: sensor.inverter_temperature
```

johannwilken, 2023-08-22:
[image]

Gorwanvfr, 2023-08-22:
Where you have yes for show daily it needs to be changed to true

hddmax, 2023-08-22:
same as here 
[image]

slipx06, 2023-08-22:
Hi. Your card configuration is incorrect. From version 2.0.0. onwards yes/no was changed to true/false. no/one/two/three/four changed to 0/1/2/3/4. Please refer to the documentation for updated examples. 

https://slipx06.github.io/sunsynk-power-flow-card/examples/sunsynk.html

hddmax, 2023-08-22:
ok i fix it :-)

MH-CODES, 2023-08-22:
Definitely not working for me either.. (can't see what is wrong in my code)
Does none become 0? 

`type: custom:sunsynk-power-flow-card
cardstyle: full
panel_mode: 'false'
large_font: 'false'
show_solar: 'true'
card_height: 400
inverter:
  modern: 'true'
  colour: grey
  autarky: power
battery:
  energy: 11000
  shutdown_soc: 20
  invert_power: 'true'
  colour: pink
  show_daily: 'true'
  animation_speed: 6
  max_power: 9000
  full_capacity: 100
  empty_capacity: 20
solar:
  colour: orange
  show_daily: 'true'
  mppts: 1
  animation_speed: 9
  max_power: 4360
  pv1_name: Garage
  pv2_name: North
  pv3_name: East
  pv4_name: West
load:
  colour: '#5fb6ad'
  show_daily: 'true'
  show_aux: 'false'
  invert_aux: 'false'
  aux_name: Generator
  aux_type: gen
  animation_speed: 8
  max_power: 8000
  additional_loads: 2
  load1_name: Geysers
  load2_name: Pool
  load1_icon: boiler
  load2_icon: mdi:pool
grid:
  colour: '#5490c2'
  show_daily_buy: 'true'
  show_daily_sell: 'false'
  no_grid_colour: '#a40013'
  show_nonessential: 'true'
  invert_grid: 'false'
  nonessential_name: Non Essential
  nonessential_icon: oven
  additional_loads: 1
  load1_name: null
  load2_name: Load 2
  load1_icon: mdi:stove
  load2_icon: mdi:ev-station
  animation_speed: 8
  max_power: 8000
entities:
  use_timer_248: switch.use_timer
  priority_load_243: switch.prioritize_load
  day_battery_charge_70: sensor.stats_daily_battery_energy_in
  day_battery_discharge_71: sensor.stats_daily_battery_energy_out
  day_load_energy_84: sensor.stats_daily_electricity_total_usage
  day_grid_import_76: sensor.stats_daily_grid_utilisation
  day_grid_export_77: sensor.sunsynk_day_grid_export
  day_pv_energy_108: sensor.stats_daily_pv_energy_generated
  inverter_voltage_154: sensor.grid_voltage
  load_frequency_192: sensor.grid_frequency
  inverter_current_164: sensor.current
  inverter_power_175: sensor.load_power
  grid_power_169: sensor.grid_power
  pv1_power_186: sensor.pv_power_1
  pv2_power_187: none
  pv3_power_188: none
  pv4_power_189: none
  pv_total: sensor.pv_power
  pv1_voltage_109: sensor.pv_voltage_1
  pv1_current_110: sensor.pv_current_1
  pv2_voltage_111: sensor.pv_voltage_2
  pv2_current_112: sensor.pv_current_2
  pv3_voltage_113: none
  pv3_current_114: none
  pv4_voltage_115: none
  pv4_current_116: none
  battery_voltage_183: sensor.battery_voltage
  battery_soc_184: sensor.battery_state_of_charge
  battery_power_190: sensor.battery_power
  battery_current_191: sensor.battery_current
  essential_power: sensor.load_power_essential
  essential_load1: sensor.combined_geyser_power
  essential_load2: sensor.pool_pump_energy_power
  nonessential_power: none
  non_essential_load1: sensor.non_essential_oven_energy_usage
  non_essential_load2: none
  grid_ct_power_172: sensor.grid_power_ct
  grid_connected_status_194: binary_sensor.grid_power_available
  inverter_status_59: sensor.device_mode
  aux_power_166: sensor.sunsynk_aux_power
  remaining_solar: sensor.solcast_forecast_remaining_today
  battery_temp_182: sensor.temperature_2
  radiator_temp_91: sensor.temperature
  dc_transformer_temp_90: sensor.sunsynk_dc_transformer_temperature
  prog1_time: sensor.sunsynk_time_slot_1
  prog1_capacity: number.sunsynk_system_mode_soc_time1
  prog1_charge: switch.sunsynk_system_mode_grid_charge_time1
  prog2_time: sensor.sunsynk_time_slot_2
  prog2_capacity: number.sunsynk_system_mode_soc_time2
  prog2_charge: switch.sunsynk_system_mode_grid_charge_time2
  prog3_time: sensor.sunsynk_time_slot_3
  prog3_capacity: number.sunsynk_system_mode_soc_time3
  prog3_charge: switch.sunsynk_system_mode_grid_charge_time3
  prog4_time: sensor.sunsynk_time_slot_4
  prog4_capacity: number.sunsynk_system_mode_soc_time4
  prog4_charge: switch.sunsynk_system_mode_grid_charge_time4
  prog5_time: sensor.sunsynk_time_slot_5
  prog5_capacity: number.sunsynk_system_mode_soc_time5
  prog5_charge: switch.sunsynk_system_mode_grid_charge_time5
  prog6_time: sensor.sunsynk_time_slot_6
  prog6_capacity: number.sunsynk_system_mode_soc_time6
  prog6_charge: switch.sunsynk_system_mode_grid_charge_time6
  energy_cost: sensor.tibber_energy_cost
`

[image]

slipx06, 2023-08-22:
You need to remove the single quotes everywhere i.e. 'false' should be false. You can also remove all those sensors that you are not using and have set to "none" 

https://slipx06.github.io/sunsynk-power-flow-card/examples/sunsynk.html

closed 2023-08-23 by johannwilken