slipx06/sunsynk-power-flow-card #67: Publish time left on battery as a sensor

bazcurtis, 2023-08-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

N/A

### Expected behaviour

If possible can you share the time remaining on the battery an a sensor. 

### Possible Solutions

Publish the time remain as a sensor. I understand that you might not want to get into the functionality as you are writing a card and might not want to get into sensors.

### Mode

Both

### Context / Reason

This would allow for charging and discharging automations to make better decisions.

slipx06, 2023-08-27:
Its outside the scope of the card, but here is the code to create you own template sensor.  Replace "20" would with your shutdown SOC and 15960 with you battery capacity 

```
- platform: template
  sensors:
    soc_battery_time_left:
      friendly_name: "Battery Depletion Seconds"
      unit_of_measurement: Seconds
      value_template: >
        {% set state = states('sensor.sunsynk_battery_power') | int %}
        {% set cap = states('sensor.battery_cap') | float %}
        {% if state == 0 -%}
        {{ ((((states('sensor.sunsynk_battery_soc') | float - 20) /100) * 15960) / (1) * 60 * 60 ) | timestamp_custom('%s', 0) }}
        {%- else -%}
        {{ ((((states('sensor.sunsynk_battery_soc') | float - 20) /100) * 15960) / (states('sensor.sunsynk_battery_power') | float) * 60 * 60 ) | timestamp_custom('%s', 0) }}
        {%- endif %}
    soc_battery_time_left_friendly:
      friendly_name: "Battery Depletion Time"
      value_template: >
        {% set state = states('sensor.sunsynk_battery_power') | int %}
        {% if state > 0 -%}
        {%- set time = states('sensor.soc_battery_time_left') | int %}
        {%- set minutes = ((time % 3600) // 60) %}
        {%- set minutes = '{} minutes'.format(minutes) if minutes > 0 else '' %}
        {%- set hours = ((time % 86400) // 3600) %}
        {%- set hours = '{} hours, '.format(hours) if hours > 0 else '' %}
        {%- set days = (time // 86400) %}
        {%- set days = '{} day, '.format(days) if days > 0 else '' %}
        {{ 'Less than 1 minute' if time < 60 else days + hours + minutes }}
        {%- else -%}
        {{ 'Charging' }}
        {%- endif %}
    markdown_discharge_time:
      friendly_name: "Markdown Discharge Time"
      value_template: >
        {% set now = as_timestamp(now()) %}
        {% set add = states('sensor.soc_battery_time_left') | int %}
        {% set future_time = now + add %}
          {{ future_time | timestamp_custom('%H:%M') }}
```

bazcurtis, 2023-08-27:
That is very fair. Thank you for sharing the code

slipx06, 2023-08-27:
Np.made some small edits to the code above

closed 2023-08-28 by slipx06

amgstone, 2023-08-30:
Good afternoon, if I have an 18 kilowatt battery, should I enter 18000?