slipx06/sunsynk-power-flow-card #68: Choose on/off grid icon colour

bazcurtis, 2023-08-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Red is off grid, green is on grid

### Expected behaviour

Green is off grid, red is on grid

### Possible Solutions

Make this an option which way you want it.

### Mode

Both

### Context / Reason

This might be the way my brain works, but being off grid is good, so should be green. Being on grid is bad, and should be red.
I understand that green is normally on and red is normally off, but in this case, I think it should be the other way around.

slipx06, 2023-08-27:
Hi. Both of there are configurable under the Grid section

<html>
<body>

Attribute | Requirement | Default | Description
-- | -- | -- | --
colour: | Optional | '#5490c2' | Changes the colour of all the grid card objects. Hex codes ('#66ff00' etc) or names (red, green, blue etc)
no_grid_colour: | Optional | '#a40013' | Changes the colour of the grid disconnected icon. Hex codes ('#66ff00' etc) or names (red, green, blue etc)

</body>
</html>

bazcurtis, 2023-08-27:
Apologies, thanks for pointing that out.

closed 2023-08-27 by bazcurtis