slipx06/sunsynk-power-flow-card #69: ON/OFF grid icon mismatch

b1lei, 2023-08-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

When grid ON it displays icon OFF grid.
When no sensor attached to the card it displays ON grid.

### Expected behaviour

icon mismatch

### Card Version

2.2

### Home Assistant Version

10.4

### Configuration

```shell
type: custom:sunsynk-power-flow-card
cardstyle: full
panel_mode: true
large_font: true
show_solar: true
card_height: ''
inverter:
  modern: true
  colour: grey
  autarky: 'no'
battery:
  energy: 60000
  shutdown_soc: 30
  invert_power: true
  show_daily: true
  animation_speed: 6
  max_power: 15000
  full_capacity: 99
  empty_capacity: 30
  colour: green
solar:
  show_daily: true
  mppts: four
  animation_speed: 9
  max_power: 17000
  pv1_name: PV1
  pv2_name: PV2
  pv3_name: PV3
  pv4_name: PV4
load:
  show_daily: true
  show_aux: false
  animation_speed: 8
  max_power: 24000
grid: 
  no_grid_colour: red
  show_daily_buy: true
  show_daily_sell: true
  show_nonessential: false
  animation_speed: 8
  max_power: 24000
  invert_grid: false
entities:
  use_timer_248: switch.inverter_1_use_timer
  day_battery_discharge_71: sensor.daily_discharge
  day_battery_charge_70: sensor.daily_charge
  day_load_energy_84: sensor.daily_load
  day_grid_import_76: sensor.daily_energy_bought
  day_grid_export_77: sensor.daily_energy_sold
  day_pv_energy_108: sensor.daily_production
  inverter_voltage_154: sensor.grid_voltage
  load_frequency_192: sensor.grid_frequency
  inverter_current_164: sensor.total_charge_current
  inverter_power_175: sensor.total_load_power
  grid_power_169: sensor.total_grid_power
  pv1_power_186: sensor.inverter_1_pv_power_1
  pv2_power_187: sensor.inverter_1_pv_power_2
  pv3_power_188: sensor.inverter_2_pv_power_1
  pv4_power_189: sensor.inverter_2_pv_power_2
  battery_voltage_183: sensor.inverter_1_battery_voltage
  battery_soc_184: sensor.battery_state_of_charge
  battery_power_190: sensor.battery_power
  battery_current_191: sensor.total_charge_current
  essential_power: sensor.total_essential_power
  grid_ct_power_172: sensor.total_grid_power
  pv1_voltage_109: sensor.inverter_1_pv_voltage_1
  pv1_current_110: sensor.inverter_1_pv_current_1
  pv2_voltage_111: sensor.inverter_1_pv_voltage_2
  pv2_current_112: sensor.inverter_1_pv_current_2
  pv3_voltage_113: sensor.inverter_2_pv_voltage_1
  pv3_current_114: sensor.inverter_2_pv_current_1
  pv4_voltage_115: sensor.inverter_2_pv_voltage_2
  pv4_current_116: sensor.inverter_2_pv_current_2
  grid_connected_status_194: sensor.deye_m01_grid_connected_status
  inverter_status_59: sensor.deye_m01_running_status
  battery_temp_182: sensor.deye_m01_battery_temperature
  radiator_temp_91: sensor.deye_m01_ac_temperature
  dc_transformer_temp_90: sensor.deye_m01_dc_temperature
```

slipx06, 2023-08-28:
Please share the output of your grid connected status sensor: `sensor.deye_m01_grid_connected_status`

The card expects a value of `on`/`off` or `1`/`0`. if no sensor is provided it will default to ON

b1lei, 2023-08-28:
id: 2709321466
integration: solarman
icon: ''
friendly_name: DEYE_M01 Grid-connected Status
   
    - name: "Grid-connected Status"
      class: ""
      state_class: ""      
      uom: ""
      scale: 1
      rule: 1
      registers: [0x00C2]
      isstr: true
      lookup: 
      -  key: 1
         value: "On-Grid"      
      -  key: 0
         value: "Off-Grid"

slipx06, 2023-08-28:
The problem is the sensor is not sending the expected data. The work around is to create a template sensor but let me look at adding this to the code

commit 5aadd724 (2023-08-28): feat: fixes #69 add solarman grid connected value
closed 2023-08-28 by slipx06