slipx06/sunsynk-power-flow-card #70: Values coming back with 3 decimal places

alanmors, 2023-08-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Irrespective of what my Display precision is set to the values come back with 3 decimal places. EG Load frequency I have set to display as 50.2 Hz and is displays that in the MQQT list but in the power flow card it displays as 50.189 Hz. This is happening no matter what browser I use and I have cleared all files. 

### Expected behaviour

I would expect to see the shortened values

### Card Version

v.2.3.0

### Home Assistant Version

2023.7.3  HAOS 10.3

jimilutz, 2023-08-29:
I think the issue is due to this PR:
https://github.com/home-assistant/core/pull/87619
I suspect these lines:"this.hass.states[config.entities.day_battery_discharge_71] "
should be of the form:
"{{ states("sensor.test, rounded=True") }}" -> 1.00
or
"{{ states.sensor.test.state_with_unit }}" -> "1.00 A"

closed 2023-08-29 by slipx06

slipx06, 2023-08-29:
Please check latest release https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v2.4.0

rickyrickuk, 2023-10-01:
Hi, I am running 2.7.0 and having a similar issue, mainly on the grid stats (Hz, Watts) but also on the Inverter Wattage (see pic)
My decimal places is set to 1 at the moment.
[image]

slipx06, 2023-10-01:
Hi. I don't apply any rounding to those power values so it displays as provided by the sensor. I will convert them to integers. 
The decimal places only applies when using the auto_scale attribute and the sensor value is greater than 999W. It displays kW with the specified number of decimal places.

The number of decimal places for Hz is the card default of 2 and at this time cannot be customised.

slipx06, 2023-10-01:
Please test [2.7.1 ](https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v2.7.1)

rickyrickuk, 2023-10-01:
> Please test [2.7.1 ](https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v2.7.1)

Thanks so much for the super fast modification, upgraded to 2.7.1 and initially looked much better but then shortly after went back to 3 decimal places.
[image]

I don't need to restart HA do I?

slipx06, 2023-10-01:
I think I know what the issue is. Have you enabled auto_scale: in your card config? If yes please remove from the inverter section  and lets see if it displays properly

rickyrickuk, 2023-10-01:
> I think I know what the issue is. Have you enabled auto_scale: in your card config? If yes please remove from the inverter section and lets see if it displays properly
> 
> Please try [2.7.2](https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v2.7.2)

Yes had auto scale set to true for inverter.

Updated to 2.7.2 and so far so good 😁