slipx06/sunsynk-power-flow-card #73: Sum up the power used with the cost

f1wade, 2023-09-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

It displays the current price, selling or buying.

### Expected behaviour

But I'd like to add the ability to sum up the daily sell with the sell cost, and the daily buy with the buy cost.

### Possible Solutions

I have had a try, but cannot get the yarn builder to work.

Would you be able to explain how to get it to compile, and or add in this option please?

### Mode

lite

### Context / Reason

lite or both.

slipx06, 2023-09-01:
Hi.  There is some guidance here https://slipx06.github.io/sunsynk-power-flow-card/contribute/devcycle.html. Have you installed all the requirements? 
If the sell and buy rate change during the day calculating the cost based on energy usage would not be a simple calculation?

f1wade, 2023-09-01:
Hi

I can't seem to get yarn to work.
On the website it says node package 4,5,6 or above 8.
I have 14.
And when I yarn install it says it wants 16 or above 18.

If I ignore that and do a yarn run build, it errors with rollup command not
found and exit code 127.

Regards
Chris wade

slipx06, 2023-09-01:
Why don't you rather setup the dev-container and it will install all the dependencies

https://slipx06.github.io/sunsynk-power-flow-card/contribute/devcontainer.html

slipx06, 2023-09-12:
Closing this for now. It's probably better handled through a template sensor but feel free to reopen if you come up with some ideas to do this

closed 2023-09-12 by slipx06