slipx06/sunsynk-power-flow-card #74: Smaller 'Liter' interface

darren-arbon, 2023-09-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

The lite version for me has much more information that I really want in an at-a-glance view

### Expected behaviour

A smaller view that is much more simplistic

### Possible Solutions

I have already forked your library and added my own there, but I was wondering if you would like me to raise a PR to add it to your package so you can see it in action.

### Mode

lite

### Context / Reason

The reason I have added a smaller version is that the current lite version has a large footprint, and does contain information that is not too important if you just want an at-a-glance view

slipx06, 2023-09-07:
Hi. Can you share a screenshot? Most things can be turned off so interested to see what you came up with.

darren-arbon, 2023-09-08:
[image]

If I can turn things off to get this kinda view I would be very happy!!

slipx06, 2023-09-08:
So by smaller, you mean the actual height of the card? You can disable sensors to create the view below

[image]

darren-arbon, 2023-09-08:
Yes, a smaller height of the card. I want this information, but with less real estate.

slipx06, 2023-09-08:
Ok. For now you can set the card height but it does make things a bit small
```
type: custom:sunsynk-power-flow-card
cardstyle: lite
show_solar: true
card_height: 315px
```
[image]

darrenarbon, 2023-09-10:
Cheers, I'll have a play and see what happens

closed 2023-09-12 by slipx06