slipx06/sunsynk-power-flow-card #75: Daily figures appearing as Absolute values, not Daily delta values.

gwynevans, 2023-09-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Hi, I’ve got an issue with my setup (which I suspect could be an issue with the source) but what I’m seeing is that all the ‘daily’ values appear as absolute values, rather than daily delta values, despite apparently being set as ‘total_increasing’.

The reason I’m doubting it’s the card is that the default device card also shows the absolute values, although the Energy dash does show the delta values!

(  I don’t know where to find/access that HA config though, as been reluctant to ‘take over’ that boards config as it’s useful having it automatically managed and don’t know if I can revert it once taken over?  Any pointers, either in HA or even the HA source, btw?)

If someone has any ideas, I’d appreciate suggestions as what to look for…

### Expected behaviour

Values such as ‘3’ rather than ‘183’ for Daily Grid usage

### Card Version

2.1.9.  (?)

### Home Assistant Version

2023.9.0

### Configuration

```shell
title: Home
views:
  - path: default_view
    title: SunSynk
    subview: false
    type: panel
    badges: []
    cards:
      - type: custom:sunsynk-power-flow-card
        cardstyle: lite
        show_solar: true
        battery:
          energy: 0
          invert_power: true
          shutdown_soc: 20
          show_daily: true
        solar:
          show_daily: true
          mppts: 2
        load:
          show_daily: true
        grid:
          show_daily_buy: true
          show_daily_sell: true
          show_nonessential: true
        entities:
          use_timer_248: switch.deye_sunsynk_sol_ark_use_timer
          priority_load_243: select.deye_sunsynk_sol_ark_energy_pattern
          inverter_voltage_154: sensor.deye_sunsynk_sol_ark_ac_output_voltage
          load_frequency_192: sensor.deye_sunsynk_sol_ark_ac_output_frequency
          inverter_current_164: sensor.sunsynk_inverter_current
          inverter_power_175: sensor.deye_sunsynk_sol_ark_load_power
          grid_connected_status_194: binary_sensor.sunsynk_grid_connected_status
          inverter_status_59: sensor.sunsynk_overall_state
          day_battery_charge_70: sensor.deye_sunsynk_sol_ark_battery_energy_in
          day_battery_discharge_71: sensor.deye_sunsynk_sol_ark_battery_energy_out
          battery_voltage_183: sensor.deye_sunsynk_sol_ark_battery_voltage
          battery_soc_184: sensor.deye_sunsynk_sol_ark_battery_state_of_charge
          battery_power_190: sensor.deye_sunsynk_sol_ark_battery_power
          battery_current_191: sensor.deye_sunsynk_sol_ark_battery_current
          grid_power_169: sensor.deye_sunsynk_sol_ark_grid_power
          day_grid_import_76: sensor.deye_sunsynk_sol_ark_grid_energy_in
          day_grid_export_77: sensor.deye_sunsynk_sol_ark_grid_energy_out
          grid_ct_power_172: sensor.deye_sunsynk_sol_ark_grid_power
          day_load_energy_84: sensor.deye_sunsynk_sol_ark_load_energy
          essential_power: none
          nonessential_power: none
          aux_power_166: sensor.sunsynk_aux_power
          day_pv_energy_108: sensor.deye_sunsynk_sol_ark_pv_energy
          pv1_power_186: sensor.deye_sunsynk_sol_ark_pv_power_1
          pv2_power_187: sensor.deye_sunsynk_sol_ark_pv_power_2
          pv1_voltage_109: sensor.deye_sunsynk_sol_ark_pv_voltage_1
          pv1_current_110: sensor.deye_sunsynk_sol_ark_pv_current_1
          pv2_voltage_111: sensor.deye_sunsynk_sol_ark_pv_voltage_2
          pv2_current_112: sensor.deye_sunsynk_sol_ark_pv_current_2
```

gwynevans, 2023-09-10:
Example: state for the Daily load sensor (sensor.deye_sunsynk_sol_ark_load_energy) from developer tools:

```
state_class: total_increasing
unit_of_measurement: kWh
device_class: energy
friendly_name: Deye/SunSynk/Sol-Ark Load energy
```

slipx06, 2023-09-10:
HI
This is not a card issue. The card will display whatever the sensor provides. It sounds like you are using the total load energy sensor which is not reset every day. You can create a new utility sensor that resets daily and use sensor.deye_sunsynk_sol_ark_load_energy as the source.  Alternatively you need to use the correct sensor from your integration that reports daily totals (if it exists)

[image]

[image]

gwynevans, 2023-09-10:
Thanks, will try that route...

closed 2023-09-10 by gwynevans