slipx06/sunsynk-power-flow-card #77: Translate

1977michael, 2023-09-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Hello, I think the card is very good. can I use the names, e.g. "daily_grid_buy": "DAILY GRID BUY",
        "daily_grid_sell": "DAILY GRID SELL", rename to another language?
If yes how?

### Expected behaviour

change the name of the labels

### Possible Solutions

other languages

### Mode

Both

### Context / Reason

Also use the card in Germany

slipx06, 2023-09-11:
Hi. Yes the card supports localisation. Here is a first attempt to add the German language. If you wouldn't mind proof reading it I can add it.

The original English definitions can be found [here](https://github.com/slipx06/sunsynk-power-flow-card/blob/8c3097384210c6f3d5ade16678690a7c9cb26ab1/src/localize/languages/en.json) 

```
{
    "common": {
        "description": "Eine Home Assistant-Karte zur Emulation des Sunsynk-Stromflusses, der auf dem Inverter-Bildschirm angezeigt wird",
        "version": "Version",
        "runtime_to": "LAUFZEIT BIS",
        "battery_floating": "BATTERIE SCHWIMMEND",
        "charge": "AUFLADEN",
        "to": "ZU",
        "daily_load": "TÄGLICHE LAST",
        "daily_aux": "TÄGLICHER HILFSANTRIEB",
        "daily_grid_buy": "TÄGLICHER NETZEINKAUF",
        "daily_grid_sell": "TÄGLICHER NETZVERKAUF",
        "daily_solar": "TÄGLICHE SOLAR",
        "daily_solar_left": "TÄGLICHE SOLAR / VERBLEIBEND HEUTE",
        "daily_charge": "TÄGLICHE AUFLADUNG",
        "daily_discharge": "TÄGLICHE ENTNAHME",
        "ratio": "Verhältnis",
        "timer_off": "Timer aus",
        "timer_on": "Timer ein",
        "priority_batt": "Batteriepriorität",
        "priority_load": "Lastpriorität",
        "nonessential_name": "Nicht essenziell",
        "aux_name": "Hilfsantrieb",
        "load1_name": "",
        "load2_name": "",
        "pv1_name": "PV1",
        "pv2_name": "PV2",
        "pv3_name": "PV3",
        "pv4_name": "PV4",
        "essential": "Essenziell"
    },
    "errors": {
        "battery": {
            "bat": "Keine Batterieattribute definiert",
            "energy": "Bitte fügen Sie das Batterieenergieattribut und den Wert in Wh hinzu, z.B. 5.32 Batterieenergie: 5320",
            "shutdown_soc": "Bitte fügen Sie das Batterie-Shutdown-SOC-Attribut und den Wert hinzu, z.B. shutdown_soc: 20",
            "full_capacity": "Die volle Kapazität muss zwischen 80 und 100 liegen",
            "empty_capacity": "Die leere Kapazität muss <= 30 sein",
            "show_daily": "Bitte fügen Sie die Tages-Batterieladung_70- und Tages-Batterieentladung_71-Attribute und die Entity-IDs hinzu"
        },
        "solar": {
            "sol": "Keine Solarattribute definiert",
            "mppts": "Bitte fügen Sie das Solar-MPPT-Attribut und den Wert hinzu; 1,2,3 oder 4 z.B. mppts: 2",
            "show_daily": "Bitte fügen Sie das Tages-PV-Energie_108-Attribut und die Entity-ID hinzu"
        },
        "grid": {
            "show_daily": "Bitte fügen Sie die Tages-Grideinspeisung_76- und Tages-Gridexport_77-Attribute und die Entity-IDs hinzu"
        },
        "essential_power": "Das Essenziell-Stromattribut ist auf 'none' gesetzt. Bitte fügen Sie das Inverter-Power_175-Attribut und die Entity-ID hinzu, damit die Karte diesen Wert berechnen kann.",
        "nonessential_power": "Das Nicht-Essenziell-Stromattribut ist auf 'none' gesetzt. Bitte fügen Sie das Grid-Power_169-Attribut und die Entity-ID hinzu, damit die Karte diesen Wert berechnen kann.",
        "missing_entity": "Bitte fügen Sie das Attribut und die Entity-ID hinzu"
    }
}

```

1977michael, 2023-09-11:
Hello, thank you very much. Where can I paste the code? I couldn't find it in File Editor.

Another problem:
I added Load_1 and load_2, but I can't find the consumers in the map.

type: custom:sunsynk-power-flow-card
cardstyle: full
show_solar: true
battery:
  energy: 7600
  shutdown_soc: 7
  show_daily: none
solar:
  show_daily: true
  mppts: 2
  pv1_name: Carport
  pv2_name: Pavillion/Haus
load:
  show_daily: true
  additional_loads: two
  load1_name: EV
  load1_icon: mdi:ev-station
  load2_name: Wärmepumpe
  load2_icon: mdi:thermometer-water
grid:
  show_daily_buy: true
  show_daily_sell: true
  show_nonessential: none
entities:
  use_timer_248: switch.sunsynk_toggle_system_timer
  priority_load_243: switch.sunsynk_toggle_priority_load
  inverter_voltage_154: sensor.sunsynk_inverter_voltage
  load_frequency_192: sensor.sunsynk_load_frequency
  inverter_current_164: sensor.sunsynk_inverter_current
  inverter_power_175: sensor.rct_power_storage_inverter_ac_power
  grid_connected_status_194: binary_sensor.sunsynk_grid_connected_status
  inverter_status_59: sensor.sunsynk_overall_state
  radiator_temp_91: sensor.rct_power_storage_db_temp1
  day_battery_charge_70: sensor.sunsynk_day_battery_charge
  day_battery_discharge_71: sensor.sunsynk_day_battery_discharge
  battery_temp_182: sensor.rct_power_storage_battery_temperature
  battery_voltage_183: sensor.rct_power_storage_battery_voltage
  battery_soc_184: sensor.rct_power_storage_battery_state_of_charge
  battery_power_190: sensor.rct_power_storage_battery_power
  battery_current_191: sensor.sunsynk_battery_current
  grid_power_169: sensor.rct_power_storage_grid_power
  day_grid_import_76: sensor.netzbezug_heute
  day_grid_export_77: sensor.einspeisung_heute
  grid_ct_power_172: sensor.rct_power_storage_grid_power
  day_load_energy_84: sensor.stromverbrauch_heute
  essential_power: sensor.rct_power_storage_consumer_power
  nonessential_power: none
  aux_power_166: sensor.sunsynk_aux_power
  day_pv_energy_108: sensor.pv_ertrag_heute
  pv1_power_186: sensor.rct_power_storage_generator_a_power
  pv2_power_187: sensor.rct_power_storage_generator_b_power
  pcarport_voltage_109: sensor.rct_power_storage_generator_a_voltage
  pv1_current_110: none
  pv2_voltage_111: none
  pv2_current_112: none
  essential_load1: sensor.ecb1_active_power_aktuelle_leistung
  essential_load2: sensor.meross_smart_switch_power_2

Mit freundlichen Grüßen

Michael Trebes
E-Mail	***@***.***
Mobil	0151 57535920

slipx06, 2023-09-11:
Change this
```
load:
  show_daily: true
  additional_loads: two
```

to

```
load:
  show_daily: true
  additional_loads: 2
```

You can either create a pull request or just paste any translation changes into this chat

slipx06, 2023-09-11:
I've added the German translation to [2.5.1](https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v2.5.1)
Please check

1977michael, 2023-09-12:
Thank you very much! The update has been completed, where can I change the language?

1977michael, 2023-09-12:
Hello
Yes, after the language was changed and returned to German, the translation in the map also works.

A question:
My inverter doesn't play AUX. I have now installed two consumers there (EV and hot water heat pump). Can I hide the "166" field?

When translating, you could use the following information (or you can adapt it yourself):

"runtime_to": "LAUFZEIT BIS",
        "battery_floating": "BATTERIE haltend",
        "charge": "AUFLADEN",
        "to": "ZU",
        "daily_load": "ENERGIE heute",
        "daily_aux": "HILFSANTRIEB heute“, or "WEITERE heute“
        "daily_grid_buy": „NETZEINKAUF heute",
        "daily_grid_sell": „NETZVERKAUF heute",
        "daily_solar": „SOLAR heute",
        "daily_solar_left": "SOLAR heute / VERBLEIBEND heute",
        "daily_charge": „AUFLADUNG heute",
        "daily_discharge": „ENTNAHME heute",
        "ratio": "Verhältnis",
        "timer_off": "Timer aus",
        "timer_on": "Timer ein",
        "priority_batt": "Batteriepriorität",
        "priority_load": "Lastpriorität",
        "nonessential_name": "Nebenverbraucher",
        "aux_name": "Hilfsantrieb",
        "load1_name": "",
        "load2_name": "",
        "pv1_name": "PV1",
        "pv2_name": "PV2",
        "pv3_name": "PV3",
        "pv4_name": "PV4",
        "essential": "Hauptverbraucher"

slipx06, 2023-09-12:
No you cant hide the field but why don't you create a template sensor that sums the two loads from the EV and hot water and use that for 166. 

Thanks for the updated translation. I will include it in the next release.

1977michael, 2023-09-12:
Yes, an template sensor is possible - that's no problem.

But:
Displaying all consumers (Essenziall, Non-Essenziall and Aux) is not possible, is it?

Essential and Aux go from the inverter.
Non-essential off the grid.
However, all consumers use the network between the inverter and the network.
Would this be solvable somehow?

slipx06, 2023-09-12:
I'm not sure of your specific use case, but the layout is defined based on the Sunsynk power flow. In this case the AUX port is bi-directional. It can be connected to a micro-inverter or generator or it can power a load, so it not only a consumption interface.

1977michael, 2023-09-12:
Thank you for your quick responses and service. I think the easiest way is to create additional entities to calculate the values ​​for me.
Thank you for this great application!!

slipx06, 2023-09-12:
Please check the de trabslation in [2.6.0](https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v2.6.0)

closed 2023-09-12 by slipx06