slipx06/sunsynk-power-flow-card #78: Title of the card is missing.

kunwarf, 2023-09-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

There is no config available to set the title of the card. Also if the name/image of the inverter can be set then it would be great.

### Expected behaviour

There should be a config for setting up the title of the card also Also if the name/image of the inverter can be set then it would be great.

### Mode

Both

### Context / Reason

I have three different inverters and battery banks installed. I am using three cards for these systems but could not differentiate between them on the dashboard so if a title field would be added then it would be great.

slipx06, 2023-09-12:
HI. There are other ways to do this on your dashboard with title card or markup card i.e. custom:mushroom-title-card.  I'll see what I can do but given the limited card space it may not be practical.

slipx06, 2023-09-12:
Hi
Added to https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v2.6.0
Update you card configuration with

```
type: custom:sunsynk-power-flow-card
cardstyle: full
show_solar: true
show_battery: true
large_font: true
title: Sunsynk Inverter
title_colour: Grey
title_size: 32px
```

[image]

closed 2023-09-12 by slipx06

kunwarf, 2023-09-13:
Thanks for your quick response.