slipx06/sunsynk-power-flow-card #79: watts is visualized incorrectly

amgstone, 2023-09-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

[image]
when I am charging the battery shows a negative value.
[image]

### Expected behaviour

should show a positive value when charging the battery

### Card Version

new

### Home Assistant Version

2023.8.4

### Configuration

```shell
type: custom:sunsynk-power-flow-card
cardstyle: lite
show_solar: true
card_height: 300px
inverter:
  modern: true
  colour: yellow
  autarky: 'no'
  model: Easun
solar:
  mppts: 2
  show_daily: true
  max_power: 7000
battery:
  energy: 18000
  shutdown_soc: 10
  show_daily: true
  max_power: 5500
  invert_power: true
  full_capacity: 90
  empty_capacity: 20
load:
  show_daily: true
  colour: '#5fb6ad'
  animation_speed: 6
  max_power: 11000
  additional_loads: 2
  load1_name: Car
  load1_icon: mdi:ev-station
  load2_name: Boiler
  load2_icon: mdi:water-boiler
grid:
  show_daily_buy: true
  show_nonessential: false
  max_power: 5500
entities:
  inverter_voltage_154: none
  load_frequency_192: none
  inverter_current_164: none
  inverter_power_175: none
  grid_power_169: sensor.smg_output_power
  battery_temp_182: sensor.bms_1_temperature_sensor
  battery_voltage_183: sensor.jk_bms_total_voltage
  battery_soc_184: sensor.jk_bms_state_of_charge
  battery_power_190: sensor.jk_bms_power
  battery_current_191: sensor.jk_bms_current
  grid_ct_power_172: sensor.smg_mains_power
  grid_connected_status_194: binary_sensor.energy_in_exist_sensor
  inverter_status_59: sensor.smg_ii_operation_mode
  pv1_power_186: sensor.smg_ii_pv_average_power
  pv2_power_187: sensor.smg_ii_2_pv_average_power
  pv1_voltage_109: none
  pv1_current_110: none
  pv2_voltage_111: none
  pv2_current_112: none
  day_pv_energy_108: sensor.soniachni_paneli
  day_battery_charge_70: sensor.bms0_charging_energy_today
  day_battery_discharge_71: sensor.bms0_discharging_energy_today
  day_load_energy_84: sensor.solar_plus_energy_minus_battery_sensor
  day_grid_import_76: sensor.spozhito_elektroenergiyi_za_sogodni
  essential_load1: sensor.car_power
  essential_load2: sensor.ten_boiler_power
```

slipx06, 2023-09-16:
Hi. This is normal behavior. When the battery is charging, the battery current and power will be negative. To show a positive value set the show_absolute attribute to true in your config.

```
battery:
  energy: 18000
  shutdown_soc: 10
  show_daily: true
  max_power: 5500
  invert_power: true
  full_capacity: 90
  empty_capacity: 20
  show_absolute: true
```

amgstone, 2023-09-16:
but when I add that value, it does not show a negative value during discharge

slipx06, 2023-09-16:
It wont because it only shows absolute (positive values).  The card cannot be configured to show a negative number for discharge.

amgstone, 2023-09-16:
I have two separate sensors for charging and discharging.

slipx06, 2023-09-16:
You need to create a template sensor that combines these. As an example the below assumes that your individual sensors return positive values i.e.

```
sensor:
 - platform: template
   sensors:
     sunsynk_battery_power:
       friendly_name: "Battery Power"
       value_template: "{{ states('sensor.battery_discharge') | float(0)) - (states('sensor.battery_charge') | float(0)   }}"
       unit_of_measurement: 'W'
```

slipx06, 2023-09-22:
Closing this for now as its not a card bug. If you have more information to share we can pick this up.

closed 2023-09-22 by slipx06