slipx06/sunsynk-power-flow-card #80: 2 phase or 3 phase support

vinaytje94, 2023-09-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Card only shows values for single phase inverters ,
[image]
I have 2 ct clamps for grid l1 l2
And have voltages for l1 and l2

### Expected behaviour

Is it possible to add a second row for the second ct sensor, or just combine both?
Also the same for the grid voltage phase A / phase B or phase AB combined, and current A B values.
I think it would be easier to implement for three phase with the option to hide the hide the parameters

### Mode

full

### Context / Reason

I have a US version of the Deye hybrid inverter. It reports 2 phase values, ct a b powers and  phase a b voltages, a b currents too

Mights001, 2023-09-20:
Hi, 

First of all, I'm really excited about the card.

I would also like if you could integrate that so that I could see all 3 phases

[image]

I just change the javascript self, but they did not work ;( 

Many Thanks in Adv.

slipx06, 2023-09-22:
Hi. Can you share the sensor data that you want displayed as I don't have a 3 phase inverter to test.

vinaytje94, 2023-09-22:
> Hi. Can you share the sensor data that you want displayed as I don't have a 3 phase inverter to test.

Hey there, 

Please see the highlighted areas
[image]

I currently have it configured like this
inverter_voltage_154: sensor.deye_voltage_l1                    /// free sensor: sensor.deye_voltage_l2
inverter_current_164: sensor.deye_current_l1                    ///  free sensor: sensor.deye_current_l2
grid_ct_power_172: sensor.deye_external_ct_l1_power     ///  free sensor: sensor.deye_external_ct_l2_power

Free sensor just means the 2nd phase, not being shown, for 3ph it wouldve been l2 and l3

I dont mind if grid ct can be combined for a total value (just like total pv).
But i think it would be handier to implement a second row below the values or besides it. You might have to increase the size of the square container to accommodate the values.

Mights001, 2023-09-22:
In my case there are 2 more values ​​for each Phase voltage and current L1-L2-L3 .
But the Modbus Addresses are not simular to the Sunsynk Invter, but that is not really importantent. 
[image]

For the load displays, it would also be great if you could include a choice between watts and kilowatts :) 

THX Mights

slipx06, 2023-09-23:
Ok. Still thinking about how we can achieve this. @Mights001 2.7.0 adds an option to display kW

Legnava89, 2023-10-06:
For the 3F inverter, I created a template in HA as a sensor.
Example for Goodwe

`  - sensor:
      - name: Napětí střídač
        unique_id: napeti
        unit_of_measurement: "V"
        state: "{{ (states('sensor.nbus_voltage') | float) + ( states('sensor.bus_voltage') | float) }}"
        icon: mdi:lightning-bolt
        
  - sensor:
      - name: Ampéry střídač
        unique_id: ampery
        unit_of_measurement: "A"
        state: "{{ (states('sensor.on_grid_l1_current') | float) + ( states('sensor.on_grid_l2_current') | float) + ( states('sensor.on_grid_l3_current') | float) }}"
        icon: mdi:lightning-bolt`

[image]

chinezbrun, 2023-10-10:
i'm interested too.

slipx06, 2023-10-10:
These are the mock-ups I have. 
It should be possible to do something like this on the lite card?

[image]

and the full card becomes a cramped but it could work 

[image]

Mights001, 2023-10-11:
Hey, that looks good. On the Lite card it might be better if you display the first voltage eache phase and than Amp and last Frequency.

slipx06, 2023-10-12:
[image]

slipx06, 2023-10-13:
Full card. You will notice that I've had to reduce decimal places show for current on the full card and voltage on the lite card to prevent overlap issues. Its a small compromise

[image]

vinaytje94, 2023-10-14:
> Full card. You will notice that I've had to reduce decimal places show for current on the full card and voltage on the lite card to prevent overlap issues. Its a small compromise
> 
> [image]

Hey slip, this looks good.
Have you added perhaps a show option for the L2/3 phases.
Is it possible to add the currents like how the voltages are listed so that the form looks uniform

vinaytje94, 2023-10-14:
Right now i have updated and adjusted mine with a micro inverter on the aux input. It looks like this
[image]

slipx06, 2023-10-14:
HI. There just isn't enough space on the card for uniformity e.g. if you have 4 mppt's 
[image]
I'm busy looking at adding L1-L3 and external CT L1-L3

slipx06, 2023-10-14:
These are the final version. Will be pushing this soon

[image]

[image]

slipx06, 2023-10-14:
See https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v2.11.0

closed 2023-10-14 by slipx06

Legnava89, 2023-10-14:
Hi, in the lite card there is a problem with the graphics of the energy_cost_sell entity

[image]

slipx06, 2023-10-14:
Thanks for catching that. Fixed in [2.11.1](https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v2.11.1)

vinaytje94, 2023-10-15:
Finally got the time to check the updates. Looks like this
[image]

slipx06, 2023-10-15:
I'm going to fix the NaN on the daily total. You can also use `inverter_current_L3:` to display your L2 current if you want to change layout a bit.