slipx06/sunsynk-power-flow-card #82: Refactor display when Aux is an input

jpettitt, 2023-10-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

When Au is a power source (eg a generate) it would be preferable to move it to a different location since the load side is somewhat cluttered.   Additionally the Ow box above the inverter is confusing in that isually is inplies power from the grid is flowing through it but it's showing 0W

[image]

### Expected behaviour

les cluttered screen,  central node not showing 0W

### Possible Solutions

Move the generator to bottow left (add a dedicated "generator" type or a flag for the aux indicating it's a generator.

Remove the redundant `0W` from the central node or reroute the connections.

### Mode

full

### Context / Reason

cluttered display, confusing value

slipx06, 2023-10-03:
Hi. A quick fix could be to change the AUX icon to represent a generator. Power flow is then represented through the animated dot.

[image]

As mentioned in your other issue those sensor values are specific to the original design intent and the Sunynk registers displayed on the system flow screen. You could hide the value by setting inverter_power_175: none or provide your own sensor to reflect the correct power flowing through your inverter.

Legnava89, 2023-10-04:
Hello, could you please add a temperature function to aux load 1 and 2?
Marked on the photo
Thank you very much for your time and answer.
[image]

slipx06, 2023-10-05:
@Legnava89 this has been added to the v2.8 https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v2.8.0

closed 2023-10-07 by slipx06