slipx06/sunsynk-power-flow-card #85: Feature request : shutdown SOC as variable input

Azylog, 2023-10-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Actually, the shutdown SOC used to calculate available time to discharge is a static entry : 
```
        battery:
          energy: 10000
          shutdown_soc: 30
```

### Expected behaviour

availability to choose between a sensor.battery_shutdown_soc as parameter or a fixed value

### Possible Solutions

I tried to used a template for the parameter like this : 
`shutdown_soc: {{ states('sensor.ess_minimum_soc_setpoint') }}` with my Victron Multiplus-II without success.

### Mode

full

### Context / Reason

Perhaps some users are modifying the shutdown SOC on a day to day basis depending on PV production and/or grid prices.
Being able to show the remaining time before shutdown with this kind of variable SOC level would be ideal.

slipx06, 2023-10-07:
Thanks for the suggestion. Will take a look

closed 2023-10-07 by slipx06

Azylog, 2023-10-08:
Wow, that was really quick, thanks a lot !
Just tried and it works like a charm : changing the minimal SOC from the display of my Victron changes the SOC and time estimation displayed on the card. Really nice !

commit 343f4239 (2023-10-07): feat: Shutdown SOC closes #85, CZ lang closes #84