slipx06/sunsynk-power-flow-card #86: aux_power_166`display negative since latest version

gonmag8, 2023-10-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Thank you very much for your work, the card is amazing, I use it everyday.

I have updated this morning to latest version, I was previously on version 2.7.1

version 2.9.1 is displaying "aux_power_166" as negative (even though I have another inverter connected to the aux port)
I have tried setting "invert_aux: false" and that does make the number positive again, but the "flow ball" is now in the wrong direction.

is there a way to invert the flow ball direction ?

Thank you for your amazing work 

### Expected behaviour

Thank you very much for your work, the card is amazing, I use it everyday.

I have updated this morning to latest version, I was previously on version 2.7.1

version 2.9.1 is displaying "aux_power_166" as negative (even though I have another inverter connected to the aux port)
I have tried setting "invert_aux: false" and that does make the number positive again, but the "flow ball" is now in the wrong direction.

is there a way to invert the flow ball direction ?

Thank you for your amazing work 

### Card Version

2.91

### Home Assistant Version

2023.10.1

slipx06, 2023-10-09:
Sorry I mixed that up 

The defaults are the other way around. Negative for input and positive for output. You can attach loads to the AUX port and in this case showing power as a positive number on the AUX load made more sense. 

There is no option to swap flow at this stage.

slipx06, 2023-10-10:
@gonmag8  I think the best approach here would be to create a show_absolute_aux attribute which will stop the negative number being displayed and you can then use the invert_aux to manage the direction of the animation.  I'll add in to the next release

gonmag8, 2023-10-10:
Sounds good, thank you very much :)

commit c5858dc9 (2023-10-10): feat: aux power as absolute value closes #86
closed 2023-10-10 by slipx06