slipx06/sunsynk-power-flow-card #88: Micro Inverter Integration

Cretanios, 2023-10-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Micro Inverter can not be displayed/integrated

### Expected behaviour

similar to Solar also the micro inverter power (and maybe frequency in off-grid mode)

### Possible Solutions

no

### Mode

lite

### Context / Reason

micro inverters are an official option to us instead of the generator option in the sunsynk/deye inverters

slipx06, 2023-10-11:
Hi, This is not displayed on the lite version of the card. You can use the AUX input on the full card to display this power

chinezbrun, 2023-10-13:
also interested on this

slipx06, 2023-10-14:
@vinaytje94 I've moved you post here

Right now i have updated and adjusted mine with a micro inverter on the aux input. It looks like this
[image]

vinaytje94, 2023-10-14:
> @vinaytje94 I've moved your post here
> 
> Right now i have updated and adjusted mine with a micro inverter on the aux input. It looks like this [image]

It looks like this when the MI is online. I have the SUN-2000WG3 MI from Deye and i used the solarman integration. 
[image]

Got it setup like this

`  show_daily_aux: true
  show_aux: true
  invert_aux: true
  aux_name: MicroInverter
  aux_type: mdi:solar-power-variant
  aux_colour: green
  aux_off_colour: gray
  aux_icon: mdi:solar-power-variant
  aux_loads: 0
  aux_load1_name: MicroInverter
  aux_load2_name: Aux load 2
  aux_load1_icon: mdi:solar-power-variant
  aux_load2_icon: mdi:stove
  animation_speed: 8
  max_power: 2000`

For the aux related inputs i used the sensors from the solarman MI integration

closed 2023-10-28 by slipx06