slipx06/sunsynk-power-flow-card #89: Versione per Telefono Cellulare

gohanp75, 2023-10-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

La versione presenta un mucchio di dati ed è ben fatta, ma sul cellulare risulta poco leggibile 

### Expected behaviour

Sarebbe bello una versione semplificata con solo i valori principali ma. Con caratteri grandi 

### Possible Solutions

Creare una versione semplificata con caratteri grandi adatta alla visualizzazione su cellulare

### Mode

lite

### Context / Reason

Avere dashboard per telefono

slipx06, 2023-10-14:
Hi. There are better card that you can use to display inverter data on small screens. I've shared my dashboard https://github.com/slipx06/Sunsynk-Home-Assistant-Dash to give you some ideas of some other cards. Have a look at the Flexible Horseshoe Card (https://github.com/AmoebeLabs/flex-horseshoe-card) It works well on mobile. . I'm not planning on building another view at this stage.

[image]
[image]

closed 2023-10-14 by slipx06

gohanp75, 2023-10-15:
a version as an attached image would have been nice.
Thank you and I hope you will consider it in the future.[image: Screenshot
2023-10-15 084538.jpg]

-- 

*PAOLI geom. Paolo*

Via delle Loze, 14

38010 Campodenno (TN)

tel 333 3174200

P.I. e C.F. 00683850226 – PLA PLA 75A06 F187Y

*Dichiarazione di esonero di responsabilità – Disclaimer*

Ai sensi del GDPR 2016/679 si precisa che questo messaggio di posta
elettronica contiene informazioni di carattere confidenziale rivolte
esclusivamente al destinatario sopra indicato.

E' vietato l'uso, la diffusione, distribuzione o riproduzione da parte di
ogni altra persona. Nel caso aveste ricevuto questo messaggio di posta
elettronica per errore, siete pregati di segnalarlo immediatamente al
mittente e distruggere quanto ricevuto (compresi i file allegati) senza
farne copia.

Qualsivoglia utilizzo non autorizzato del contenuto di questo messaggio
costituisce violazione dell'obbligo di non prendere cognizione della
corrispondenza tra altri soggetti, salvo più grave illecito, ed espone il
responsabile alle relative conseguenze.

slipx06, 2023-10-15:
@gohanp75 I cant see your screenshot?

gohanp75, 2023-10-15:
I am attaching an image of my cell phone, another image is my elaboration
to understand how I would like to see and replace the current meters (which
resemble yours)
I would also like to point out that there is no translation into Italian,
if I can help you let me know.
I have several friends who would like a dashboard like this.
Thanks again and I hope you will consider it.[image: Screenshot 2023-10-15
084538.jpg][image:
Screenshot_20231015_130754_io.homeassistant.companion.android.jpg]

-- 

*PAOLI geom. Paolo*

Via delle Loze, 14

38010 Campodenno (TN)

tel 333 3174200

P.I. e C.F. 00683850226 – PLA PLA 75A06 F187Y

*Dichiarazione di esonero di responsabilità – Disclaimer*

Ai sensi del GDPR 2016/679 si precisa che questo messaggio di posta
elettronica contiene informazioni di carattere confidenziale rivolte
esclusivamente al destinatario sopra indicato.

E' vietato l'uso, la diffusione, distribuzione o riproduzione da parte di
ogni altra persona. Nel caso aveste ricevuto questo messaggio di posta
elettronica per errore, siete pregati di segnalarlo immediatamente al
mittente e distruggere quanto ricevuto (compresi i file allegati) senza
farne copia.

Qualsivoglia utilizzo non autorizzato del contenuto di questo messaggio
costituisce violazione dell'obbligo di non prendere cognizione della
corrispondenza tra altri soggetti, salvo più grave illecito, ed espone il
responsabile alle relative conseguenze.

slipx06, 2023-10-15:
The emails are not attaching your images. If you can try upload into the thread directly https://github.com/slipx06/sunsynk-power-flow-card/issues/89#issuecomment-1763313628

Thanks for offering to help with the Italian translation. Can you proof read the content below

```
{
    "common": {
        "description": "Una scheda Home Assistant per emulare il flusso di potenza Sunsynk visualizzato sullo schermo dell'inverter",
        "version": "Versione",
        "runtime_to": "DURATA A",
        "battery_floating": "BATTERY FLOATING",
        "charge": "CARICA",
        "to": "A",
        "daily_load": "CARICO GIORNALIERO",
        "daily_aux": "AUX GIORNALIERO",
        "daily_grid_buy": "ACQUISTO GIORNALIERO DALLA RETE",
        "daily_grid_sell": "VENDITA GIORNALIERA ALLA RETE",
        "daily_solar": "ENERGIA SOLARE GIORNALIERA",
        "daily_solar_left": "ENERGIA SOLARE GIORNALIERA / RIMASTA OGGI",
        "daily_charge": "CARICA GIORNALIERA",
        "daily_discharge": "SCARICA GIORNALIERA",
        "ratio": "Rapporto",
        "timer_off": "Spegnimento timer",
        "timer_on": "Accensione timer",
        "priority_batt": "Priorità batteria",
        "priority_load": "Priorità carico",
        "nonessential_name": "Non Essenziale",
        "aux_name": "Ausiliario",
        "load1_name": "",
        "load2_name": "",
        "pv1_name": "PV1",
        "pv2_name": "PV2",
        "pv3_name": "PV3",
        "pv4_name": "PV4",
        "essential": "Essenziale"
    },
    "errors": {
        "battery": {
            "bat": "Nessun attributo della batteria definito",
            "energy": "Si prega di includere l'attributo dell'energia della batteria e il valore in Wh, ad esempio 5,32 Energia della batteria: 5320",
            "shutdown_soc": "Si prega di includere l'attributo di spegnimento della batteria e il valore, ad esempio spegnimento_soc: 20",
            "full_capacity": "La capacità massima deve essere compresa tra 80 e 100",
            "empty_capacity": "La capacità vuota deve essere <= 30",
            "show_daily": "Si prega di includere gli attributi day_battery_charge_70 e day_battery_discharge_71 e gli ID delle entità"
        },
        "solar": {
            "sol": "Nessun attributo solare definito",
            "mppts": "Si prega di includere l'attributo solare mppts e il valore; 1,2,3 o 4, ad esempio mppts: 2",
            "show_daily": "Si prega di includere l'attributo day_pv_energy_108 e l'ID dell'entità"
        },
        "grid": {
            "show_daily": "Si prega di includere gli attributi day_grid_import_76 e day_grid_export_77 e gli ID delle entità"
        },
        "essential_power": "L'attributo essential_power è impostato su none. Si prega di includere l'attributo inverter_power_175 e l'ID dell'entità in modo che la scheda possa calcolare questo valore.",
        "nonessential_power": "L'attributo nonessential_power è impostato su none. Si prega di includere l'attributo grid_power_169 e l'ID dell'entità in modo che la scheda possa calcolare questo valore.",
        "missing_entity": "Si prega di includere l'attributo e l'ID dell'entità"
    }
}

```

gohanp75, 2023-10-15:
[image]

gohanp75, 2023-10-15:
[image]

gohanp75, 2023-10-15:
I hope you see it now

gohanp75, 2023-10-15:
{
    "common": {
        "description": "Una scheda Home Assistant per emulare il flusso di potenza Sunsynk visualizzato sullo schermo dell'inverter",
        "version": "Versione",
        "runtime_to": "DURATA AL",
        "battery_floating": "BATTERY CARICA",
        "charge": "CARICA",
        "to": "A",
        "daily_load": "CONSUMO GIORNALIERO",
        "daily_aux": "AUX GIORNALIERO",
        "daily_grid_buy": "CONSUMO GIORNALIERO RETE",
        "daily_grid_sell": "VENDITA GIORNALIERA RETE",
        "daily_solar": "ENERGIA SOLARE GIORNALIERA",
        "daily_solar_left": "ENERGIA SOLARE GIORNALIERA / RIMASTA OGGI",
        "daily_charge": "CARICA GIORNALIERA",
        "daily_discharge": "SCARICA GIORNALIERA",
        "ratio": "Rapporto",
        "timer_off": "Spegnimento timer",
        "timer_on": "Accensione timer",
        "priority_batt": "Priorità batteria",
        "priority_load": "Priorità carico",
        "nonessential_name": "Non Essenziale",
        "aux_name": "Ausiliario",
        "load1_name": "",
        "load2_name": "",
        "pv1_name": "PV1",
        "pv2_name": "PV2",
        "pv3_name": "PV3",
        "pv4_name": "PV4",
        "essential": "Essenziale"
    },
    "errors": {
        "battery": {
            "bat": "Nessun attributo della batteria definito",
            "energy": "Si prega di includere l'attributo dell'energia della batteria e il valore in Wh, ad esempio 5,32 Energia della batteria: 5320",
            "shutdown_soc": "Si prega di includere l'attributo di spegnimento della batteria e il valore, ad esempio spegnimento_soc: 20",
            "full_capacity": "La capacità massima deve essere compresa tra 80 e 100",
            "empty_capacity": "La capacità vuota deve essere <= 30",
            "show_daily": "Si prega di includere gli attributi day_battery_charge_70 e day_battery_discharge_71 e gli ID delle entità"
        },
        "solar": {
            "sol": "Nessun attributo solare definito",
            "mppts": "Si prega di includere l'attributo solare mppts e il valore; 1,2,3 o 4, ad esempio mppts: 2",
            "show_daily": "Si prega di includere l'attributo day_pv_energy_108 e l'ID dell'entità"
        },
        "grid": {
            "show_daily": "Si prega di includere gli attributi day_grid_import_76 e day_grid_export_77 e gli ID delle entità"
        },
        "essential_power": "L'attributo essential_power è impostato su none. Si prega di includere l'attributo inverter_power_175 e l'ID dell'entità in modo che la scheda possa calcolare questo valore.",
        "nonessential_power": "L'attributo nonessential_power è impostato su none. Si prega di includere l'attributo grid_power_169 e l'ID dell'entità in modo che la scheda possa calcolare questo valore.",
        "missing_entity": "Si prega di includere l'attributo e l'ID dell'entità"
    }
}

gohanp75, 2023-10-15:
I corrected as best I could, I still have a doubt on a couple of definitions
"daily_solar_left":  "..... / RIMASTA OGGI"
"daily_aux":         "AUX GIORNALIERO"

what do you mean by these values?

slipx06, 2023-10-15:
"daily_solar_left": "..... / RIMASTA OGGI"  This is the amount of Solar Energy left for the day based on weather
[image]

"daily_aux": "AUX GIORNALIERO"
The daily energy for the AUX input/output on the inverter
[image]

gohanp75, 2023-10-16:
{
"common": {
"description": "Una scheda Home Assistant per emulare il flusso di potenza Sunsynk visualizzato sullo schermo dell'inverter",
"version": "Versione",
"runtime_to": "DURATA AL",
"battery_floating" : "BATTERY CARICA",
"charge": "CARICA",
"to": "A",
"daily_load": "CONSUMO OGGI",
"daily_aux": "AUX OGGI",
"daily_grid_buy": "PRELIEVO OGGI DA RETE",
"daily_grid_sell": "CESSIONE OGGI IN RETE",
"daily_solar": "ENERGIA SOLARE OGGI",
"daily_solar_left": "ENERGIA SOLARE OGGI / STIMA RIMANENTE",
"daily_charge": "CARICA OGGI",
"daily_discharge": "SCARICA OGGI" ,
"ratio": "Rapporto",
"timer_off": "Spegnimento timer",
"timer_on": "Accensione timer",
"priority_batt": "Priorità batteria",
"priority_load": "Priorità carico",
"nonessential_name": " Non Essenziale",
"aux_name": "Ausiliario",
"load1_name": "",
"load2_name": "",
"pv1_name": "PV1",
"pv2_name": "PV2",
"pv3_name": "PV3",
"pv4_name": "PV4",
"essential": "Essenziale"
},
"errors": {
"battery": {
"bat": "Nessun attributo della batteria definito",
"energy": "Si prega di includere l' attributo dell'energia della batteria e il valore in Wh, ad esempio 5,32 Energia della batteria: 5320",
"shutdown_soc": "Si prega di includere l'attributo di spegnimento della batteria e il valore, ad esempio spegnimento_soc: 20" ,
"full_capacity": "La capacità massima deve essere compresa tra 80 e 100",
"empty_capacity": "La capacità vuota deve essere <= 30",
"show_daily": "Si prega di includere gli attributi day_battery_charge_70 e day_battery_discharge_71 e gli ID delle entità"
},
"solar": {
"sol": "Nessun attributo solare definito",
"mppts": "Si prega di includere l'attributo solare mppts e il valore; 1,2,3 o 4,
ad esempio mppts: 2", "show_daily": "Si prega di includere l'attributo day_pv_energy_108 e l'ID dell'entità"
},
"grid": {
"show_daily": "Si prega di includere gli attributi day_grid_import_76 e day_grid_export_77 e gli ID delle entità"
},
"essential_power": "L'attributo essenziali_power è impostato su none. Si prega di includere l'attributo inverter_power_175 e l'ID dell'entità in modo che la scheda possa calcolare questo valore.",
"nonessential_power": "L'attributo nonessential_power è impostato su none. Si prega di includere l'attributo grid_power_169 e l'ID dell'entità in modo che la scheda possa calcolare questo valore.",
"missing_entity": "Si prega di includere l'attributo e l'ID dell'entità"
}
}