slipx06/sunsynk-power-flow-card #91: Interface does not translate - HA language not detected

adnovea, 2023-10-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

I have installed all the update for a couple of months and though the language translation issue will be solved. 
Now I guess it's time to report this issue.
My HA interface is in French but the add-on labels are in English.
Looking at the SRC folder, there is a file for french translation and inside the sunsynk-power-flow-card.js file I can found the french strings.
I have no clue what could be wrong.

### Expected behaviour

Card strings should translate

### Card Version

2.10.0

### Home Assistant Version

2023.10.0

### Configuration

```shell
type: custom:sunsynk-power-flow-card
cardstyle: full
panel_mode: false
large_font: false
show_solar: true
show_battery: true
inverter:
  modern: true
  colour: lightgrey
  autarky: power
  model: sunsynk
  auto_scale: true
battery:
  energy: 0
  shutdown_soc: 20
  invert_power: true
  colour: '#f06292'
  show_daily: true
  animation_speed: 6
  max_power: 2000
  full_capacity: 100
  empty_capacity: 0
  show_absolute: true
  auto_scale: false
solar:
  colour: orange
  show_daily: true
  mppts: 2
  animation_speed: 9
  max_power: 4920
  pv1_name: Ouest
  pv2_name: Est
  auto_scale: false
load:
  colour: '#5fb6ad'
  show_daily: true
  show_daily_aux: true
  show_aux: true
  invert_aux: false
  animation_speed: 8
  max_power: 5000
  aux_name: Surplus (V.E.)
  show_absolute_aux: true
  aux_type: default
  aux_colour: lightblue
  aux_off_colour: lightgrey
  aux_loads: 0
  aux_load1_name: Véhicule
  aux_load1_icon: mdi:car-electric-outline
  aux_load2_name: Piscine
  aux_load2_icon: mdi:pool
  additional_loads: 0
  load1_name: Secouru
  load1_icon: mdi:power-plug-battery-outline
  auto_scale: false
grid:
  colour: '#488fc2'
  no_grid_colour: rgb(192,0,0)
  show_daily_buy: true
  show_daily_sell: true
  show_nonessential: true
  nonessential_icon: default
  nonessential_name: Gros consommateurs
  additional_loads: 0
  load1_name: Non essentielles
  load1_icon: mdi:washing-machine
  invert_grid: false
  animation_speed: 8
  max_power: 6000
  auto_scale: false
entities:
```

slipx06, 2023-10-13:
Hi. This does work if you explicitly set you HA Dashboard language
[image]

The problem is the translation does not look good and needs some work to fix alignment

[image]

adnovea, 2023-10-13:
Hi splix06,

Thanks for your quick reply.
I had the language set under Parametres > System > General

[image]

but the add-on always displays English strings

[image]

Any idea what could be wrong ?

French is wordy so we need to find some shortcuts.
It could be something like :
```
    "common": {
        "description": "Une carte Home Assistant pour émuler les flux d'énergie affichés à l'écran des onduleurs Sunsynk ",
        "version": "Version",
        "runtime_to": "AVANT D'ATTEINDRE",
        "battery_floating": "BATTERIE FLOTTANTE",
        "charge": "CHARGE",
        "to": "À",
        "daily_load": "CHARGE DU JOUR",
        "daily_aux": "CHARGE AUX DU JOUR",
        "daily_grid_buy": "ACHAT CE JOUR",
        "daily_grid_sell": "VENTE CE JOUR",
        "daily_solar": "PRODUCTION DU JOUR",
        "daily_solar_left": "PRODUCTION / RESTANT (JOUR)",
        "daily_charge": "CHARGE DU JOUR",
        "daily_discharge": "DÉCHARGE DU JOUR",
        "ratio": "Ratio",
        "timer_off": "Prog. M.",
        "timer_on": "Prog. A.",
        "priority_batt": "Batterie en 1er",
        "priority_load": "Charges en 1er",
        "nonessential_name": "Non Essentielles",
        "aux_name": "Auxiliaire",
        "load1_name": "",
        "load2_name": "",
        "pv1_name": "PV1",
        "pv2_name": "PV2",
        "pv3_name": "PV3",
        "pv4_name": "PV4",
        "essential": "Essentielles"
    },

```

slipx06, 2023-10-13:
Thanks I will update with your suggestions.
You need to set the language under your profile for the card translation to work.

adnovea, 2023-10-13:
Thanks for the hint but it was already the case.
[image]

So what I did was to select Deutsche and the translation worked,.
I selected back French and it worked.
I logout and login again but the card is back in English. 
Look like you must re-select language each time you login.

slipx06, 2023-10-13:
Thanks for confirming the solution. That might be a HA bug or maybe your browser is caching some files. I'm not sure. You can try clear your browser cache or double check in a new incognito window........or just don't log out :)

adnovea, 2023-10-13:
Hi, I already did the test using an other browser (Edge instead of FF)  and this is how I found the strange behavior.
Clearing the browser cache reset the language selection. You must stay logged after changing the user's language.
Many thanks for you nice add-on. I run it for a DEYE 5K.

closed 2023-10-13 by slipx06

adnovea, 2023-10-15:
Oups, timer_off/on inverted in my French translation.
I suggest the following to solve this issue. THX
```
        "timer_off": "Prog. Off.",
        "timer_on": "Prog. On",
```
Where "Autarky" can be translated to "Autoconso" ?
Could decimals of "energy_cost_buy" be selected 0-2 ?

slipx06, 2023-10-15:
Thanks I will fix that. I want to leave the term Autarky as it is a concept of self sufficiency that could get lost in translation. I will add an additional attribute to the Grid section, `energy_cost_decimals: 2` so you can customise the number of decimal places for energy cost buy and sell

slipx06, 2023-10-17:
> Oups, timer_off/on inverted in my French translation. I suggest the following to solve this issue. THX
> 
> ```
>         "timer_off": "Prog. Off.",
>         "timer_on": "Prog. On",
> ```
> 
> Where "Autarky" can be translated to "Autoconso" ? Could decimals of "energy_cost_buy" be selected 0-2 ?

@adnovea  Fixed and added in v2.12.0

adnovea, 2023-10-17:
Thanks for you quick support and for your work.

tigers75, 2023-10-17:
I think this should be added to the main page FAQ, it's not the expected behaviour and it's very rare that people track down a closed issue on GitHub

slipx06, 2023-10-18:
When you log a bug report I do ask that you please check open and closed issues

[image]

If more people trip over this I'll add it to the main page.

tigers75, 2023-10-18:
Well, I am aware and I did search, but couldn't find it. So will many that will come after me.
Also since this is the only add-in that doesn't get translated properly I believe the issue is NOT resolved, and should be investigated more and not closed.
Maybe it happens only to who updated the card, and not on fresh installs. Anyway I don't think it's normal.

slipx06, 2023-10-26:
@tigers75 I've made some updates to the translation logic.  Can you test [3.1.0](https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v3.1.0) if it solves the issue

jzielke84, 2024-02-03:
I also encountered this issue and I am staying logged in. In Fully Kiosk Browser on my Huawei Tab, the language in my profile is set to German but the cad shows English, while on my Desktop, the card shows German. So this is not resolved.

slipx06, 2024-02-03:
@jzielke84 can you try changing your language to English and then back top German on you tablet.

jzielke84, 2024-02-03:
@slipx06 This worked, but wth 😳  Why did all other cards show up correctly? Doing this would have never occoured to me. Thank you!