slipx06/sunsynk-power-flow-card #92: 3-phase system error

Richardrsb, 2023-10-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

In a 3-phase system the L1-power and L1-Voltage are not shown.
L2 and L3 are correct.
The L1 Voltage is not shown at all. L2 and L3 are correct.

### Expected behaviour

Correct representation of the power and voltage of L1
Please note i have re-configured the settings to work with a Deye 20k hybrid.

[image]

### Card Version

Latest as per okt-15-2023

### Home Assistant Version

Latest as per okt-15-2023

### Configuration

```shell
type: custom:sunsynk-power-flow-card
cardstyle: full
card_height: 440px
panel_mode: false
large_font: true
title: Deye 20kW Inverter
title_colour: grey
title_size: 24px
show_solar: true
show_battery: true
decimal_places: 2
inverter:
  modern: false
  colour: '#bbbbbb'
  autarky: energy
  auto_scale: true
  three_phase: true
battery:
  energy: 60000
  shutdown_soc: 10
  invert_power: true
  colour: pink
  show_daily: true
  animation_speed: 6
  max_power: 20000
  full_capacity: 90
  empty_capacity: 10
  show_absolute: false
  auto_scale: true
solar:
  colour: '#55ffaa'
  show_daily: true
  mppts: 2
  animation_speed: 6
  max_power: 20000
  pv1_name: PV1
  pv2_name: PV2
  pv3_name: East
  pv4_name: West
  auto_scale: true
load:
  colour: '#5faaad'
  show_daily: true
  show_daily_aux: true
  show_aux: false
  invert_aux: false
  show_absolute_aux: false
  aux_name: Generator
  aux_type: gen
  aux_colour: green
  aux_off_colour: red
  aux_loads: 2
  aux_load1_name: Aux load 1
  aux_load2_name: Aux load 2
  aux_load1_icon: mdi:heat-pump-outline
  aux_load2_icon: mdi:air-conditioner
  animation_speed: 6
  max_power: 8000
  additional_loads: 2
  load1_name: Heatpump
  load2_name: Airco
  load1_icon: mdi:heat-pump-outline
  load2_icon: mdi:air-conditioner
  auto_scale: true
grid:
  colour: '#5490ff'
  show_daily_buy: true
  show_daily_sell: true
  no_grid_colour: '#a40013'
  show_nonessential: true
  invert_grid: false
  nonessential_name: Non Essential
  nonessential_icon: mdi:grid-large
  additional_loads: 2
  load1_name: Heatpump
  load2_name: Airco
  load1_icon: mdi:heat-pump
  load2_icon: mdi:air-conditioner
  animation_speed: 6
  max_power: 20000
  auto_scale: true
entities:
  use_timer_248: switch.use_timer
  priority_load_243: switch.sunsynk_toggle_priority_load
  day_battery_charge_70: sensor.battery_energy_in
  day_battery_discharge_71: sensor.battery_energy_out
  day_load_energy_84: sensor.load_energy
  day_grid_import_76: sensor.grid_energy_in
  day_grid_export_77: sensor.grid_energy_out
  day_pv_energy_108: sensor.pv_energy
  day_aux_energy: sensor.load_energy
  inverter_voltage_L1: sensor.p1_meter_3c39e72b7eaa_active_voltage_l1
  inverter_voltage_L2: sensor.p1_meter_3c39e72b7eaa_active_voltage_l2
  inverter_voltage_L3: sensor.p1_meter_3c39e72b7eaa_active_voltage_l3
  load_frequency_192: sensor.grid_frequency
  inverter_current_164: sensor.auxiliary_load_current_essential
  inverter_power_175: sensor.load_power_essential
  grid_power_169: sensor.grid_power
  pv1_power_186: sensor.pv_power_1
  pv2_power_187: sensor.pv_power_2
  pv_total: sensor.pv_power
  pv1_voltage_109: sensor.pv_voltage_1
  pv1_current_110: sensor.pv_current_1
  pv2_voltage_111: sensor.pv_voltage_2
  pv2_current_112: sensor.pv_current_2
  battery_voltage_183: sensor.battery_voltage
  battery_soc_184: sensor.battery_state_of_charge
  battery_power_190: sensor.battery_power
  battery_current_191: sensor.battery_current
  essential_power: sensor.load_power_essential
  essential_load1: sensor.egauge_heatpump_use
  essential_load2: sensor.egauge_airco_use
  nonessential_power: sensor.load_power
  non_essential_load1: sensor.egauge_heatpump_use
  non_essential_load2: sensor.egauge_airco_use
  grid_connected_status_194: binary_sensor.sunsynk_grid_connected_status
  inverter_status_59: sensor.sunsynk_overall_state
  aux_power_166: sensor.auxiliary_load_power
  aux_connected_status: binary_sensor.sunsynk_aux_connected_status
  remaining_solar: sensor.solcast_forecast_remaining_today
  battery_temp_182: sensor.battery_temperature
  radiator_temp_91: sensor.temperature
  dc_transformer_temp_90: sensor.temperature
  prog1_time: sensor.sunsynk_time_slot_1
  prog1_capacity: number.capacity_point_1
  prog1_charge: switch.grid_charge_point_1
  prog2_time: sensor.sunsynk_time_slot_2
  prog2_capacity: number.capacity_point_2
  prog2_charge: switch.grid_charge_point_2
  prog3_time: sensor.sunsynk_time_slot_3
  prog3_capacity: number.capacity_point_3
  prog3_charge: switch.grid_charge_point_3
  prog4_time: sensor.sunsynk_time_slot_4
  prog4_capacity: number.capacity_point_4
  prog4_charge: switch.grid_charge_point_4
  prog5_time: sensor.sunsynk_time_slot_5
  prog5_capacity: number.capacity_point_5
  prog5_charge: switch.grid_charge_point_5
  prog6_time: sensor.sunsynk_time_slot_6
  prog6_capacity: number.capacity_point_6
  prog6_charge: switch.grid_charge_point_6
  solar_sell_247: switch.solar_export_when_battery_full
  inverter_current_L1: sensor.p1_meter_3c39e72b7eaa_active_current_l1
  inverter_current_L2: sensor.p1_meter_3c39e72b7eaa_active_current_l2
  inverter_current_L3: sensor.p1_meter_3c39e72b7eaa_active_current_l3
  #grid_ct_power_172: sensor.grid_power_ct
  grid_ct_power_L1: sensor.grid_power_ct_1
  grid_ct_power_L2: sensor.grid_power_ct_2
  grid_ct_power_L3: sensor.grid_power_ct_3
```

### Relevant log output

```shell
No log.
```

slipx06, 2023-10-17:
Hi. For L1 voltage and power you need to use the following attributes:

inverter_voltage_154:
grid_ct_power_172:

The sensor names were not adjusted for backward compatibly with the single phase setup. I'll try make this clearer in the documentation

Richardrsb, 2023-10-17:
Excellent! I was just busy to find this myself. I have set now:

....
  inverter_voltage_154: sensor.p1_meter_3c39e72b7eaa_active_voltage_l1
  inverter_voltage_L2: sensor.p1_meter_3c39e72b7eaa_active_voltage_l2
  inverter_voltage_L3: sensor.p1_meter_3c39e72b7eaa_active_voltage_l3
....
  grid_ct_power_172: sensor.grid_power_ct_1
  grid_ct_power_L2: sensor.grid_power_ct_2
  grid_ct_power_L3: sensor.grid_power_ct_3
....

Off course the P1 readings are equal to the inverter readings. Just to be clear.

Thanks! This is excellent!

closed 2023-10-17 by slipx06