slipx06/sunsynk-power-flow-card #97: load is calculated incorrectly and invert_grid

seebaer1976, 2023-10-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

The calculation of load is incorrect, the inverter power and the feed into the grid are added together instead of deducted. And when purchasing from the grid, the value is deducted instead of added to it.

Feeding into the grid is a positive number, drawing from the grid is a negative number.

[image]

and with the mains and battery I can't get the power values and the animation to match. If the values are correct, the point is wrong, the point is correct, the values are not correct.

I tried everything with invert_grid, invert_power and show_absolute, but nothing gave the right result.

### Expected behaviour

maybe there is a solution

### Card Version

v2.12.0

### Home Assistant Version

2023.10.3

### Configuration

```shell
type: custom:sunsynk-power-flow-card
cardstyle: full
card_height: 440px
panel_mode: false
large_font: true
title: Huawei SUN2000-(3KTL-10KTL)-M1
title_colour: grey
title_size: 24px
show_solar: true
show_battery: true
decimal_places: 2
inverter:
  animation_speed: 8
  max_power: 10
  autarky: power
  modern: true
  three_phase: true
  auto_scale: true
  colour: white
battery:
  energy: 10000
  shutdown_soc: 2
  animation_speed: 5
  max_power: 5
  full_capacity: 90
  empty_capacity: 10
  invert_power: true
  show_absolute: false
  show_daily: true
  auto_scale: true
  colour: green
solar:
  mppts: 2
  pv1_name: Ost
  pv2_name: West
  animation_speed: 8
  max_power: 10
  auto_scale: true
  show_daily: true
  colour: orange
load:
  animation_speed: 8
  max_power: 8
  additional_loads: 1
  load1_name: Pool
  load1_icon: mdi:pool
  show_daily: true
  auto_scale: true
  colour: blue
grid:
  show_daily_buy: true
  show_daily_sell: true
  show_nonessential: false
  auto_scale: true
  invert_grid: true
  animation_speed: 8
  max_power: 8
  colour: yellow
entities:
  day_battery_charge_70: sensor.pv_batterie_day_charge
  day_battery_discharge_71: sensor.pv_batterie_day_discharge
  day_grid_import_76: sensor.zahler_am550_bezug_tag
  day_grid_export_77: sensor.zahler_am550_einspeisung_tag
  day_load_energy_84: sensor.hausverbrauch_tag
  day_pv_energy_108: sensor.pv_ertrag_tag
  battery_voltage_183: sensor.pv_batterie_bus_voltage
  battery_soc_184: sensor.pv_batterie_state_of_capacity
  battery_power_190: sensor.pv_batterie_charge_discharge_power
  battery_current_191: sensor.pv_batterie_bus_current
  dc_transformer_temp_90: sensor.wechselrichter_internal_temperature
  essential_load1: sensor.pool_watt
  grid_connected_status_194: none sensor.wechselrichter_off_grid_status
  grid_power_169: sensor.power_meter_active_power
  grid_ct_power_172: sensor.power_meter_phase_a_active_power
  grid_ct_power_L2: sensor.power_meter_phase_b_active_power
  grid_ct_power_L3: sensor.power_meter_phase_c_active_power
  inverter_voltage_154: sensor.wechselrichter_phase_a_voltage
  inverter_current_164: sensor.wechselrichter_phase_a_current
  inverter_current_L2: sensor.wechselrichter_phase_b_current
  inverter_voltage_L2: sensor.wechselrichter_phase_b_voltage
  inverter_current_L3: sensor.wechselrichter_phase_c_current
  inverter_voltage_L3: sensor.wechselrichter_phase_b_voltage
  inverter_power_175: sensor.wechselrichter_active_power
  load_frequency_192: sensor.power_meter_frequency
  pv_total: sensor.pv_produktion_aktuell_mit_effizienzverlust
  pv1_voltage_109: sensor.wechselrichter_pv_1_voltage
  pv1_current_110: sensor.wechselrichter_pv_1_current
  pv2_voltage_111: sensor.wechselrichter_pv_2_voltage
  pv2_current_112: sensor.wechselrichter_pv_2_current
  pv1_power_186: sensor.pv_string_1_watt
  pv2_power_187: sensor.pv_string_2_watt
```

slipx06, 2023-10-18:
Hi. The formula to calculate essential power (load) is for the single phase inverter and based on the Sunsynk registers. (`essential = inverter_power_175 + grid_power_169 - aux_power_166`). Since you are using a 3 phase setup and a different inverter (possibly different measurement points) you should rather provide your own load sensor in the card config. You can add the following entity:

`essential_power: <your sensor here>`

seebaer1976, 2023-10-29:
problem solved
I simply put on the appropriate sensors.
It's a bit more complicated, but it works

closed 2023-10-29 by seebaer1976