slipx06/sunsynk-power-flow-card #100: 3 phase inverter

adiweps, 2023-10-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

grid_power_169

How can I have multiple entries that give me a total?

Example, I need that block to show the sum of the 3 below for the line item grid_power_169

sensor.solarman_load_l1_power
sensor.solarman_load_l2_power
sensor.solarman_load_l3_power

### Expected behaviour

sensor.solarman_load_l1_power = 200
sensor.solarman_load_l2_power = 200
sensor.solarman_load_l3_power = 200

the block that displays grid_power_169 should = 600

### Mode

lite

### Context / Reason

Allows you to see consumption of all three phases towards the house.

slipx06, 2023-10-25:
Hi. The load power for the three phases should equal the `essential_power` shown on the lite card. The card does not currently calculate this value correctly when using three phase. If your integration does not provide you with a sensor for total load power then the recommendation for now is to create a template sensor that sums these and use that for your essential_power sesnor

adiweps, 2023-10-25:
Magic… thanx!!!

adiweps, 2023-10-25:
Any guidance on getting a block to show the current rate of battery discharge in the pink?

[image]

like on the app?

[image]

Its a Deye 12Kva Inverter

slipx06, 2023-10-25:
Are you referring to the remaining run time or charge time? Not sure what you mean by rate of battery discharge.

adiweps, 2023-10-25:
If you look at the 2nd image, it shows you what the current draw is on the battery to power the house.

833 discharge / output of current load
718 draw / current load

have a block in the pink to show the 833 in image 1

slipx06, 2023-10-25:
I prefer to group all the battery info.  It just works better with the card layout and available space. You can hide the battery voltage and current sensors by setting them to `none`.  Might relook at this in the future

[image]

adiweps, 2023-10-26:
Any chance you can share your code view for your image? Will see what I can make work for us?

slipx06, 2023-10-26:
You will need to adjust the following SVG elements

https://github.com/slipx06/sunsynk-power-flow-card/blob/1852f09006680496d7a8fff9e72a78cc7581c20c/src/index.ts#L1137
https://github.com/slipx06/sunsynk-power-flow-card/blob/1852f09006680496d7a8fff9e72a78cc7581c20c/src/index.ts#L1201-L1206
https://github.com/slipx06/sunsynk-power-flow-card/blob/1852f09006680496d7a8fff9e72a78cc7581c20c/src/index.ts#L1286-L1289
https://github.com/slipx06/sunsynk-power-flow-card/blob/1852f09006680496d7a8fff9e72a78cc7581c20c/src/index.ts#L1350-L1364

adiweps, 2023-11-01:
Could you possibly let me know where to edit the SVG elements?

slipx06, 2023-11-01:
The card itself is an SVG container with a number elements e.g. rect, paths and other SVG's. I've tried to link those that are relevant to the battery information above. For example the svg below is for the battery image:

`<svg xmlns="http://www.w3.org/2000/svg" id="bat-high" x="232.5" y="325.5" width="78.75" height="78.75" preserveAspectRatio="none" opacity="${parseInt(stateObj12.state) >= bat_full ? 1 : 0}" viewBox="0 0 24 24"> <path class="${config.show_battery === false ? 'st12' : ''}" fill="${battery_colour}" d="M 12 20 H 4 V 6 h 8 L 12 20 m 0.67 -16 H 11 V 2 H 5 v 2 H 3.33 C 2.6 4 2 4.6 2 5.33 v 15.34 C 2 21.4 2.6 22 3.33 22 h 9.34 c 0.74 0 1.33 -0.59 1.33 -1.33 V 5.33 C 14 4.6 13.4 4 12.67 4 M 11 16 H 5 v 3 h 6 v -3 m 0 -9 H 5 v 3 h 6 V 7 m 0 4.5 H 5 v 3 h 6 v -3 h -3 h 3"/></svg>`

You and use any SVG path editor or SVG editor to adjust and then update the code.

closed 2023-11-06 by adiweps