slipx06/sunsynk-power-flow-card #106: shutdown_soc: - using sensor no longer works after v3.2.0 upgrade

RhysMcW, 2023-10-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

on v2.13.1 the battery shutdown_soc: value worked when set as a sensor, now after upgrading to 3.2.0 this nor longer works, looks to require a numeric value being directly defined.
with battery_soc: set to use a sensor (shutdown_soc: sensor.sunsynk_battery_capacity_shutdown) now displays the following next to the battery
NaN min RUNTIME TO sensor.sunsynk_battery_capacity_shutdown% @NaN:NaN
setting "shutdown_soc: 20" correctly displays the time estimation...

### Expected behaviour

display time estimation, but using the sensor or direct value

### Possible Solutions

re-introduce the previous code that used either the direct defined value or get the value from a defined sensor
shutdown_soc: sensor.sunsynk_battery_capacity_shutdown

### Mode

Both

### Context / Reason

if the shutdown SoC is changed on the inverter and having defined shutdown_soc using a sensor it would automatically update on the display without having to edit the card

slipx06, 2023-10-27:
Hi. That code has not changed and is still working for me. Have you checked in developer tools that your sensor `sensor.sunsynk_battery_capacity_shutdown` is providing the correct output?

RhysMcW, 2023-10-27:
Hi
Apologies, looks like I just had to wait a little longer for the sensor to get a value again after the OS update that happened too...

closed 2023-10-27 by RhysMcW