slipx06/sunsynk-power-flow-card #110: Not showing the power from grid

taHC81, 2023-10-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

I set the valid sensor for grid_power_169: sensor.electricity_usage_w , but it doesn't show it within the card.   day_grid_import_76: sensor.daily_grid_energy_sum is reported correctly.

### Expected behaviour

I'd expect to see the actual power taken from a grid.

### Card Version

3.3.0

### Home Assistant Version

core-2023.10.5

taHC81, 2023-10-29:
Two pictures...
[image]
[image]

slipx06, 2023-10-29:
Hi. The correct sensor is `grid_ct_power_172: sensor.electricity_usage_w`

taHC81, 2023-10-29:
Thanks, now working fine. But it wasn't clear from the configuration docu.

closed 2023-10-29 by slipx06