slipx06/sunsynk-power-flow-card #112: CS language - correction

mamo-nick, 2023-10-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

"description": "Karta Home Assistant pro emulaci toku energie, který je zobrazen na obrazovce měniče Sunsynk", 
"exporting": "Prodej",
"importing": "Nákup",

### Expected behaviour

"description": "Karta Home Assistant pro emulaci toku energie, který je zobrazen na displeji měniče Sunsynk", 
"exporting": "Export",
"importing": "Import",

### Card Version

3.5.1

### Home Assistant Version

2023.10.3

slipx06, 2023-10-31:
Hi. Are these terms generally not translated for CS? 
Interested to hear some other views before updating. @Legnava89 ?

mamo-nick, 2023-10-31:
Hi,
This terms are generraly translated to czech language as 
exporting - vývoz
importing - dovoz. 
But export/import are domesticated words in czech as other languages.

Legnava89 translated this as
Prodej / Nákup, it means in czech llanguage
Sell/Buy.

This attribute expresses power flow direction, not bussines relationship with energy company, so correct translation is export/import.
For example  if I use virtual battery, I transfer energy to/from virtual battery. No sell or buy (Prodej/Nákup).
Thanks

FajnB, 2023-11-01:
Hi @mamo-nick 

**TLDR:** I also think it should be renamed to Export/Import

**The long version:**
You are technically right and wrong at the same time.

It might seem counterintuitive to translate export/import to sell/buy but for most people this is pretty much right and how they use their PV.
The term export/import is here to generalise the idea of this flow of electricity to and from grid
As you said, some people might have virtual battery and this suits them better.
In my view as long as you are connected to the grid you are technically selling the electricity even though your energy provider "saves" your Wh to virtual battery. (It is way more complicated than that but this should be sufficient explanation)

That being said.
I agree with you that it should be generalised.

In Czech language we do have quite heavily used Anglicism of export/import. In fact the exact same words.
So in sake of everything already being referred to as export/import (in the apps and other PV SW) I'd call it the Export/Import in Czech as well.

Legnava89, 2023-11-01:
these expressions are translated correctly. this one person wants them in the english language

slipx06, 2023-11-02:
Export and Import have multiple meanings in the English language as well, but it terms of energy flow through the Inverter commonly used . What if we used sending and receiving to represent energy flow instead of Sell/Buy:

```
exporting: "Odesílání" 
importing: "Příjem" 
```

Legnava89, 2023-11-02:
Ok i agree.
exporting: "Odesílání" 
importing: "Příjem"

mamo-nick, 2023-11-02:
Hi @slipx06 

Export and Import also have multiple meanings in all the languages ​​you've translated.

Why did you translate export/import as export/import (in local form) in all languages, but only in CS this way?

I don't agree, but it's your decision as the author of the integration and I respect it.

And why is the term Standby not translated?

The imperfect translation is the very reason why I would like to be able to have this card in English even though I have a dashboard in Czech.
See https://github.com/slipx06/sunsync-power-flow-card/issues/113

Anyway, thanks for your work.

FajnB, 2023-11-02:
@Legnava89 
> These expressions are translated correctly. This one person wants them in the english language I buy or sell electricity from the distribution network.
> 
> ```
>     "exporting": "Prodej",
>     "importing": "Nákup",
> ```
> 
> Export / import in the Czech Republic is used as a trade in goods from the Czech Republic to the whole world. In the domestic market, there is only buying and selling. Most people use buying and selling at spot prices in the market. So, I wouldn't change anything about the translation.

The longer I am looking at your response the more I realise that I probably got carried away by professional deformation.
Don't get me wrong, I still think it is for best to generalise it, but using anglicism wasn't probably the best idea from me.
I even wrote why I feel this way above.

@mamo-nick 
> And why is the term Standby not translated?

Sorry, but with this I know for a fact, that this is non issue in Czech.
Yes there are two possibilities (Standby [or] Pohotovostní režim / V pohotovosti) but these hold the same weight in terms of usage.
Yes, some people 45+ are more likely to use (pohotovostní režim), but most of the people I know from my family and friend either do not even call this state anything like that (Off or sleeping) or do use the two mentioned above interchangeably.
And even some manuals, written in Czech, these days state a "stand-by" mode.
So I think this is irrelevant and not worth changing.

@slipx06 

> Export and Import have multiple meanings in the English language as well, but it terms of energy flow through the Inverter commonly used . What if we used sending and receiving to represent energy flow instead of Sell/Buy:
> ```
> exporting: "Odesílání" 
> importing: "Příjem" 
> ```

About your suggestion.  **I** mostly **agree with it.**
But there is something about exporting as "Odesílání" that quite doesn't hit the spot.
Maybe my colleagues have something to add. Like I don't know, something like "Dodávání" but this isn't right either, because this is dependent on a context as it can be used for both import and export.
I don't know. Maybe I am just lost and nitpicking about small things. :D

Anyways, as @mamo-nick said. It is your project and I believe that now you have enough information to decide for what is the best.
Thank you for this project. It is a life saver.

Legnava89, 2023-11-02:
@FajnB  
thank you for your opinion.
I was based on the entities daily_grid_buy and daily_grid_sell from the grid, where the daily purchase and sale are already clearly written.
Everyone understands it without problems. Jens, do you mind exporting/importing translated purchases/sales?
In the Facebook group, everyone is happy to have such a wonderful card in the Czech language, but only one person still has a problem.

slipx06, 2023-11-02:
Who would have thought translating into Czech could be so contentious :smiley: Anyway thank you for all the input and discussion. The opinions are divided but I'd like to find a middle ground that would suite the majority of users. Is the general consensus that we settle on the below? If I get thumbs up I'll make the change

```
exporting: "Odesílání" 
importing: "Příjem" 
```

@mamo-nick as a workaround to keep those two items as Export and Import a quick fix/hack is to edit the js file that can be found in www>community>sunsynk-power-flow-card and just do a find and replace and reload the card.  You would just need to remember to do this with each update but it literally a few minutes effort. @Legnava89 you can do the same if you prefer to use selling/buying etc.  

[image]

closed 2023-11-03 by slipx06

slipx06, 2023-11-05:
Hi All. Just a heads up that this is going to change based on the info provided in #119 
It will more accurately reflect the work mode provided by register 179. Please make sure your sensor.work_mode_code returns the following:

```
    goodwe: {
        standby: { states: ['0', 'Wait Mode'], color: 'blue', message: localize('common.standby') },
        normal: { states: ['1', '2', 'Normal (On-Grid)', 'Normal (Off-Grid)'], color: 'green', message: localize('common.normal') },
        fault: { states: ['3', 'Fault Mode'], color: 'red', message: localize('common.fault') },
        flash: { states: ['4', 'Flash Mode'], color: 'yellow', message: localize('common.flash') },
        check: { states: ['5', 'Check Mode'], color: 'orange', message: localize('common.check') },
    }    
```

commit 6d6f00f9 (2023-11-03): feat: closes #115 , locale: closes #112